nightscout/AndroidAPS #230: 2.8.1.1 - Virtual Pump driver

peterholubek, 2021-01-13:
On the Master App, in Virtual Pump module, when I switch to concrete Pump, it hold it not for long time, when I see after 12 hours in Virtual Pump settings, It is again on default "Generic virtual pump".

Build version: 2.8.1.1

Thank You for fixing.

MilosKozak, 2021-01-13:
Master app with virtual pump?

peterholubek, 2021-01-13:
No, NSClient and Virtual pump setting, for correct showing boluses in NSclient App.Odoslané z môjho zariadenia Galaxy
-------- Pôvodná správa --------Od: Milos Kozak <notifications@github.com> Dátum: 13. 1. 2021  11:37  (GMT+01:00) Komu: nightscout/AndroidAPS <AndroidAPS@noreply.github.com> Kópia: peterholubek <peterholubek2013@gmail.com>, Author <author@noreply.github.com> Predmet: Re: [nightscout/AndroidAPS] [2.8.1.1] - Virtual Pump driver (#230) 
Master app with virtual pump?

MilosKozak, 2021-01-18:
I just tested it with DanaRS and it works like expected.
PM with you address of NS

closed 2021-01-18 by MilosKozak