nightscout/AndroidAPS #239: Omnipod "Pump" (Orange/EmaLink) battery age not shown (Not urgent)

robertrub, 2021-01-17:
I have an Omnipod with an Orangelink. 

AAPS 2.8.1.1 shows the battery level of the Orangelink, which is working :)

But, if I note (on the Actions page) _Pump battery change_ with a **date** (today or before), the information is noted on the care portal but not shown in the actions page. 

Orangelink battery is NOT the pump battery but it is the only necessary battery information needed with Omnipod. 

As Orangelink is in beta testing, having the age of the battery shown will help giving them the necessary information too.

[image]
[image]

vanelsberg, 2021-01-20:
Confirming.
Seeing this too.

andyrozman, 2021-10-16:
OrangeLink is the only RL/RL-clone that we can read battery from... 

And display you have shown, would show age of battery in pump (your case Pod). We can't read battery from Pod, so in this case age shouldn't be shown at all... 

Displaying Battery level of OL, wouldn't be really correct here... Especially since we don't know how long we have used the battery on OL.

robertrub, 2021-10-17:
Sorry, I don't agree on battery age for OL having no meaning. My OL arrives at 50% in about 1 month. Even though it says "50%", when I measure the battery voltages, they are on the low side and more importantly, OL has more disconnect/reconnect problems. I use the battery age in NS to know when I'll need to change the batteries.  As we already have this information, it would be nice to show it ;)

As for DASH, no battery info is needed and, if possible, the "battery info" line should not be shown at all.

vanelsberg, 2021-10-17:
@robertrub Agree that for OL battery age would make some sense. This could then be covered by the user option to enable the field?

(Just my personal view: Remembering battery **percentage** of OL on the EROS tab was quite accurate I mostly monitored this to determine battery health. Personally I had no interest in battery **age** - other then when testing OL to get and idea on how many days batteries would last - used pen-and-paper for that ;-)

Andries-Smit, 2022-03-08:
For Omnipod EROS, the age should be working, when the "Change battery pump" action is performed. Change action button is only available for EROS when enabled through the preferences. For Dash battery section in action and overview status light will be hidden completely https://github.com/nightscout/AndroidAPS/pull/1328

closed 2022-03-08 by robertrub