nightscout/AndroidAPS #242: [2.8.1.1] 3dots Actions Preferences greyed out when on ACT Tab

chrisbev, 2021-01-19:
[2.8.1.1] 3 dots top right  Actions Preferences greyed out when on ACT Tab.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

AdrianLxM, 2021-01-19:
The Actions Plugin has no preferences.

closed 2021-01-19 by AdrianLxM

chrisbev, 2021-01-19:
If I go to the Home Tab then 3 dots gives 'Actions Preferences' and I can then set colours/levels for status lights etc.?

chrisbev, 2021-01-20:
Root cause seems to be confusion for me over Home preferences that change name. Hence what I thought were 'ACT preferences' are 'Home Tab preferences' but for me the 3dots said 'Actions preferences' see #244.