nightscout/AndroidAPS #246: Everyday shutdown of AAPS at 11pm

Mondelman, 2021-01-20:
Every day, around 11 pm, AAPS 2.8.1.1 is no longer working and it has to be started again.

I can include the log files, but it's generating around 100 files per day, being 5 MB each. Most of the information I've been able to see is related to Automation and the profile changes that I use.
Thanks for commenting.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

Mondelman, 2021-01-20:
[logexport.zip](https://github.com/nightscout/AndroidAPS/files/5841784/logexport.zip)
[AndroidAPS_LOG_1611139225101.log.zip](https://github.com/nightscout/AndroidAPS/files/5841786/AndroidAPS_LOG_1611139225101.log.zip)

MilosKozak, 2021-01-20:
search FB group for PlayProtect

closed 2021-01-20 by MilosKozak