nightscout/AndroidAPS #248: 2.8.1.1  Objectif 8. Autosens is ON but it is not

robertrub, 2021-01-21:
I started objectif 8 tonight. It says activation of Autosens. I check in settings and Autosens in ON (as it was before...) but on the OpenSMB page, it said Autosens NOT activated.

I had to turn OFF and then ON the Autosens in settings in order to activate it in OpenSMB listing.

I think that Autosens slider should be OFF till the end of Objectif 7 and then be turned ON (but really ON, not just the slider on ON but inactivated...)

AdrianLxM, 2021-01-24:
Do you have a screenshot of the OpenAPS SMB fragment please?
It might be a coincidence that toggeling the setting did have an effect.

How long after starting the Autosens objective did you see this? Did you have a new BG value in between triggering a loop run?

robertrub, 2021-01-24:
Hi @AdrianLxM 
Sorry, no screen capture :( Usually I make them automatically but seems like I did not this time. 
The AutoSens slider was _ON_ since I started using the AAPS (2.7 main) with an Omnipod simulation. I can't recall if **I** put it _ON_ (even if it wasn't actively _ON_ ) or if the default setup was set with the slider on _ON_.

Then I received 1 RL and 1 OL and installed 2.8 main and then 2.8.1.1 main. As soon as I reached obj 8, I checked the AutoSens slider and it was still _ON_ but the OpenAPS SMB page said AutoSens was not activated (or was _OFF_, can't recall the exact terms either).  

I went back to the slider and turned it _OFF_ and then _ON_ and when back to the OpenAPS SMB page and saw that I had much more data (all those = and + and - etc). 

All this back and forth took less than 2-3 minutes after the activation of obj 8. It is possible that there was a BG reading from my Libre1+MM2 sent to GLIMP and from there sent to NS and AAPS. 

Hope this helps you a bit.

AdrianLxM, 2021-01-24:
Then it is highly likely that the OpenAPS SMB tab was still showing the data from the last loop run before activating the objective.
I tried to reproduce it and with the next loop it resolves as expected.

I'll close this for now. Please feel free to open it again if this happens again and doesn't go away after one more loop run.

Thanks for reporting these observations! It helps a lot.

closed 2021-01-24 by AdrianLxM

robertrub, 2021-01-25:
Hi

OK, it is totally possible I was too eager and rapid ;)

But, don't you think that the AutoSense slider should on OFF till it can 
be used (objectif 8) ?

Robert