nightscout/AndroidAPS #253: Manual AAPS delivered bolus not picked up in treatments, may not be in IOB

tim2000s, 2021-01-22:
Reporting bugs
--------------
21.05 - manual bolus administered via AAPS and showing in pump history not showing in treatments or uploaded to Nightscout. Not clear that it's included in IOB. IOB may be undercounted. 

At the time AAPS issued a basal not updated followed by a bolus not delivered alert. 

Pump in use is Medtronic 722. Logs uploaded. 
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

tim2000s, 2021-01-22:
Ran OpenAPS rig against the pump and confirmed that the bolus was missing from IOB.

andyrozman, 2021-01-22:
That is weird... When you administer bolus from AAPS this information is added to treatments, unless something went wrong... Even if it wasn't added to treatment, if it was delivered with pump, it should have been picked up from history... 

Can you please provide logs @tim2000s

MilosKozak, 2021-01-23:
we don't have logs

tim2000s, 2021-01-23:
That's odd. They were sent through last night.

tim2000s, 2021-01-23:
I've resent the logs through just now. To the usual logs@androidaps.org address.

AdrianLxM, 2021-01-23:
@tim2000s androidaps.org email addresses do no longer work.

tim2000s, 2021-01-24:
> @tim2000s androidaps.org email addresses do no longer work.

That's the address that AAPS used from within the system...

StephenBrown2, 2021-01-28:
What email should be used? I have sent several logs via the Maintenance menu, now I think they must have gone nowhere, though I haven't gotten any bounce responses.

closed 2021-11-10 by MilosKozak