nightscout/AndroidAPS #254: 2.8.1.1 Reboot phone and a strange counting message

robertrub, 2021-01-23:
I rebooted my Samsung S9 Android v10 phone. When the phone had started, I looked in AAPS 2.8.1.1 and there was a strange counting message. I'm not sure what it is connecting to (I suppose Orangelink but not sure) but the message was very strange and uncomprehensible.

[image]

MilosKozak, 2021-01-23:
it's connection to the pump/rileylink. just counter. if translation is wrong fix it in crowdin

closed 2021-01-23 by MilosKozak

robertrub, 2021-01-23:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAEgAAABICAYAAABV7bNHAAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAAArSURBVHic7cEBDQAAAMKg909tDjegAAAAAAAAAAAAAAAAAAAAAAAAAAA+DFFIAAEctgHwAAAAAElFTkSuQmCC###24x24:true###][Image de pistage][https://github.com/notifications/beacon/ASITUKSME6F26DYSRXAQXVTS3KDOZA5CNFSM4WPQF472YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFWTI2EQ.gif]

Hi Milos

After your first message asking me to correct things in Crowdin, I had asked to join but that was a week ago and I have not heard anything from them yet. I'm waiting at the door...

I'd like to thank you personally for all the work, time and brain cells you (and the team of course) put in AAPS, it is a magnificent piece of work. I'm a type 1 since '84 and been very keen on a strict control of the beast. This is the first time in 36 years that a big part of the burden has been lifted.

You can call on me for help, testing or finance. My coding days are decades away but testing and logics are still in my books :)

Thanks again and best regards.

Robert

----------------------------------------