nightscout/AndroidAPS #262: AAPS stopped working, same issue as 2.8.1.1

petervoogt, 2021-01-25:
This morning AAPS stopped working, I installed build 2.8.2 yesterday becouse 2.8.1.1 had the same problem. According to NS AAPS last connection was @5.22

[AndroidAPS._2021-01-25_05-24-31_.0.zip](https://github.com/nightscout/AndroidAPS/files/5864891/AndroidAPS._2021-01-25_05-24-31_.0.zip)

greetings

MilosKozak, 2021-01-25:
what do you mean by stopped working? i see in logs working between 3AM and 6AM

petervoogt, 2021-01-27:
A few day ago I installed 2.8.1.1. the morning of the (24th) I was woken up by a high BG alert (10.0+). Then going to my dougther's room finding the app being shut down..... In NS it is visible that no attemt was made to correct rising BG's. NS plugin "openAPS" shows last connection 5.22u .That morning updated to 2.8.2 and the next day the exact same problem. 
Now reinstalled a very old version and no problems.
I am sure I selected android 9.0 when building both of the app's, the same version as the phone.

petervoogt, 2021-01-31:
yesterday I build the APK again, but this time updated android studio to Android Studio 4.1.2 first

Build #AI-201.8743.12.41.7042882, built on December 20, 2020
Runtime version: 1.8.0_242-release-1644-b01 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
Windows 10 10.0
GC: ParNew, ConcurrentMarkSweep
Memory: 1237M
Cores: 8
Registry: ide.new.welcome.screen.force=true, external.system.auto.import.disabled=true
Non-Bundled Plugins: com.thoughtworks.gauge, com.intellij.marketplace, org.jetbrains.kotlin 

Selected android 8.1, 9.0 and 10.0. Also updated Xdrip+ to the latest version.
Last night everything was working perfectly. My conclusion, considering logs working, has to be an overloaded phone closing the app. (Cubot kingkong mini running on stock 9.0)

Do you agree with this conclusion?

closed 2021-02-01 by MilosKozak