nightscout/AndroidAPS #264: [2.8.1.1] not reading reservoir from Dana RS consistently

JohnKitching, 2021-01-25:
Two hours after changing set and reservoir AAPS is showing 13 units in pump. 
Tap Bluetooth icon on Dana tab and it updates to show 278. 
Will email logs now. 

[image]

[image]

AdrianLxM, 2021-02-14:
Can you reproduce this on 2.8.2?

JohnKitching, 2021-02-14:
It's not something I can recreate but I'll watch out for it thanks. I'll let you know either way. 

Out of interest, does AAPS calculate or read the reservoir level or is it a combination of both? How often is it supposed to read from the pump?

JohnKitching, 2021-02-14:
I've just had a thought ... I let AAPS pick up the set change from the pump. Could that event maybe not be triggering a reservoir level read when it should?

guydavies, 2021-02-14:
I had a couple of similar instances. In one, it picked up after the next
read. In the other, I don't know what happened because the pump hadn't
recognised the refill and prime. I had to reprime up to 7u to get it
recognised and then the app picked it up.

JohnKitching, 2021-02-14:
I've never had AAPS fail to pick up the set change but often fail to read the reservoir after (maybe always but I'm not sure). 
I haven't noticed in 2.8.2 if it happens. I only updated last week (sorry)
I do prime and cannula fill on the pump, not via AAPS if that makes a difference.

guydavies, 2021-02-14:
I also prime and fill from the pump. It was probably user error in my case.
Not able to say what happened for you 😊

JohnKitching, 2021-02-16:
I can confirm it's still happening in 2.8.2
I just changed cannula and insulin. 
AAPS picked up the change and shows CAGE and IAGE as 0d0h. 
After about 12 mins and two BGs reservoir is still showing 33u but the pump shows 287u

nshtg, 2021-03-27:
To force a re-read of the pump reservoir I manually force close aaps in the android settings and restart it. Then it will start a query and display the correct insulin amount.

JohnKitching, 2021-03-28:
> To force a re-read of the pump reservoir I manually force close aaps in the android settings and restart it. Then it will start a query and display the correct insulin amount.

You don't have to do that. Tapping the Bluetooth icon on the Dana tab forces the app to read the Dana and display the correct value.

MilosKozak, 2021-10-08:
full status is read every 30 min to save battery.  you can force full read by pressing BT icon
anyway I changed code to read reservoir status more often in dev branch

closed 2021-10-08 by MilosKozak