nightscout/AndroidAPS #265: [AAPS 2.8.1.1 and 2.8.2] Password necessary to export the configuration

winni67, 2021-01-25:
Before and after installing AAPS 2.8.2 over my AAPS 2.8.1.1 installation, I have to enter a master password. But I have disabled this master password for the configuration export.
Kind regards.

MilosKozak, 2021-01-26:
i don't understand you question/issue properly

rICTx-T1D, 2021-01-26:
@winni67 In all version they can remember,  you need a master password to export or import settings. The password encrypt and protect your data's, for example the nightscout account....

Where you see an issue here?

winni67, 2021-01-27:
@rICTx-T1D Thats not true. To export an unencrypted setting in the last versions (2.7), it was not necessary to enter a master password. For me, that was the way to see, if I export unencrypted or encrypted. This is a change in the behavior and perhaps an issue.

winni67, 2021-01-27:
@MilosKozak To enter the master password for an unencrypted setting is new to AAPS 2.8. In AAPS 2.7 it was not necessary. Is this a new feature or is this an issue? Now I can‘t see if the exported file is encrypted or unencrypted, because the password is now necessary for both export versions.

MilosKozak, 2021-01-28:
it's not possible to export with old unencrypted format anymore.
you have to provide password

closed 2021-01-31 by MilosKozak