nightscout/AndroidAPS #267: INFO: Pumpe nicht erreichbar

Harrypatz, 2021-01-26:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

Harrypatz, 2021-01-26:
Is with me also so, was already at 2.7 and 2.8 and also at 2.8.1, that my phone starts to beep without what is displayed. Only by the sound I know that the pump can not be reached. Can I then only end in which I click in settings the app AAPS end force.
Now with 2.8.2 to
09:55
@MilosKozak 

[AndroidAPS_LOG_1611651378771.log.zip](https://github.com/nightscout/AndroidAPS/files/5872317/AndroidAPS_LOG_1611651378771.log.zip)

AdrianLxM, 2021-01-26:
Could you please describe your setup and your settings?
Could you please attach a screenshot of the Overview (Home) in AAPS when this happens?

Harrypatz, 2021-01-26:
What do you want to know from the settings, the list is long, or do you want to have a backup file.
On the "Home" view is nothing more to see, is as always.
In this case it was after a bolus delivery with a wizzard.

MilosKozak, 2021-01-26:
i see the alarm is comming from NS Announcement.
You can turn it off if you don't need it
Or try to scroll down overview page if the notification appears

Harrypatz, 2021-01-26:
@MilosKozak 

NS alarms are switched off

StephenBrown2, 2021-01-26:
Need to turn on "Use system notifications" under "Local alerts". Can filter the full preference list for "notification" and you'll find it right away.
[image]

MilosKozak, 2021-01-26:
NS alarms and NS announcements ?

Harrypatz, 2021-02-10:
Yes, unfortunately it occurs again and again and more frequently

AdrianLxM, 2021-02-14:
Does it still happen if you fully disable NSClient in the ConfigBuilder?

Harrypatz, 2021-03-27:
@AdrianLxM  Yes, unfortunately. It's annoying, especially when you're on the road. Another problem is that the smartphone gives an alarm after the bolus has been delivered that no insulin could be delivered, in this case with a proper error message and can also be turned off, but in the smartwatch the bolus runs correctly and is also ended there normally. (Sony SW3)

closed 2021-10-01 by MilosKozak