nightscout/AndroidAPS #268: [enhancement] New action "notice" or "sport-entry" in automation

rICTx-T1D, 2021-01-26:
Hi folks,

I wish me a new feature in automation. Please make it possible to set a action for making a notice or sport-entry in care-portal. So I can track times where I drive in my car when I connected with bluetooth or time when I make sport with hearth sensor (BT-device too).

Thanks.

guydavies, 2021-01-26:
This is already possible.  I do it for driving in my car.  You have to have a specific start/stop or have it timed and as long as the test still passes, it resets back again for the specified time.

Rgds,

Guy

rICTx-T1D, 2021-01-26:
HOw action must I choose to write a notice in careportal and NS?

guydavies, 2021-01-26:
Ah, sorry, I misunderstood your requirement.  I thought you wanted to take an existing action based on connectivity to BT.

I perform the kind of task you’re asking with Automate (the Android app).  I have a few rules in Automate that automatically see when I’m active (walk, run, cycle, etc) and calculate when it’s finished and push an activity record to NS.   I then have the NS upload only disabled (careful here folks :) ) and I can see the events in AAPS.

Automate can also identify when a particular BT device is attached to the phone, so you could use that as a trigger.  I use that to activate my phone as a hotspot which is used by my M5StickC in the car to display my BG.

Rgds,

Guy

rICTx-T1D, 2021-01-26:
@guydavies 
Thanks for your answer, but I don't like install additional software or tools to perform my request. AAPS detect my BT device successfully only a action set is missing to make the entry in careportal/ NS. I hope anyone can build it.

guydavies, 2021-01-26:
Understood. But that's your only choice at the moment. It's a pretty niche request so I don't see it being something that would be added quickly.

closed 2024-12-04 by MilosKozak