nightscout/AndroidAPS #270: Ns client the battery don't update is ok all in master

adrianagomezdiaz, 2021-01-28:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

Philoul, 2021-01-28:
Can you explain what is your problem please?
I don't understand your issue...

adrianagomezdiaz, 2021-01-31:
[image]

MilosKozak, 2021-02-01:
NSCleint synchronizes only 2 days to the past. is this value updated when you enter new Battery change on master phone?

adrianagomezdiaz, 2021-02-01:
Yes in master is ok. It change. But in ns client no

adrianagomezdiaz, 2021-02-02:
Ns client is really very bad with 2.8.2. Grm