nightscout/AndroidAPS #271: Synch AAPS and NS Client not working

Garweho, 2021-01-28:
I got error message: io.socket.engineio.client.EngineIOException: xhr poll error. AAPS doesn't upload data to my Nightscout site anymore. The app and site itself are working well. Xdrip sends glucose data to Nightscout and AAPS and if I input carbs or insulin via xdrip, it shows up on Nightscout site. But AAPS is not synching with Nightscout. So if I insert carbs or insulin in AAPS, it's going nowhere. I use AAPS version 2.8.2, but got this error message already in the previous version. But before that it worked. 

[image]
[image]
[image]

MilosKozak, 2021-01-28:
where is you NS hosted? based on the message AAPS is loosing connection to the server until it's get paused
it's NS or internet issue

Garweho, 2021-01-28:
it's hosted on my own pi server. strange it worked fine for several months, but started just now. don't remember I made any changed to server. do you know what this error message means or what problem the server could have?

rapocalypse, 2021-01-29:
I'm seeing the same problem with a friend, that runs the NS on Heroku. Xdrip is sending BS readings to NS on Heroku, but AAPS isn't connecting to it, and pausing the client.

Even after restarting the smartphone or just the NSClient, for about 2 minutes the connection seams to be ok, but right after it goes on pause again.

Garweho, 2021-01-30:
OK, tried old Version ( AAPS 2.7.0) which definately worked before. Having same problem now. So I guess it must really have something to do with internet or server/NS.

closed 2021-02-01 by MilosKozak