nightscout/AndroidAPS #273: Basal+Bolus=TDD

JimZam, 2021-01-29:
Good morning. I don't know if it's an issue, just a question about Daily Dosis. In Actions/Daily Dose AAPS reports the Total Daily Dose for several days as the sum of Bolus + Basal. But if I add all the bolus reported in Treatments or uploaded to NightScout it's not the same as the Bolus reported in Daily Dose. I've checked for several days and it's never the same. Hos does AAPS calculate the Daily Bolus? Which is the correct one?
Thank you and regards

AdrianLxM, 2021-02-09:
This is the pump history. It counts whatever the pump counts into daily Basal/Bolus/TDD

closed 2021-02-09 by AdrianLxM

szantos, 2021-02-22:
@AdrianLxM 
This info should be displayed with a hint which calculation is the correct one.

Nightscout report shows a third value:

[image]

In AAPS it would be sufficient to display only the correct values, and Nightscout should match it. The current system causes a bit of confusion.

Please consider reopening.