nightscout/AndroidAPS #275: Alarm modifications (Omnipod (?), RileyLink)

SugrNichi, 2021-01-30:
When AAPS cannot update the basal profile (e.g. after profile switches), or if insulin delivery is suspended, the app plays a really drilling, dramatic alarm sound that will not stop until it is manually turned off. There are many occasions in life when one cannot simply do that. Examples include showers, in certain work situations, when speaking in front of people etc.
During the day, a simple push notification is more than enough for many people. Phones are used ubiquitously these days, so it is hard to miss a push notification during waking hours. Night is a different story, of course.

I would be extremely grateful if the following features were implemented into AAPS:

- The option to turn off audible alarms during certain hours (e.g. morning till bedtime)
- The option to change the alarm sound (e.g. to one gentle alarm sound every 5 minutes until acknowledged, or a custom sound)

I am using AAPS 2.8.2 and the Omnipod driver with a RileyLink. If I am not mistaken, these alarms have been around since one of the later 2.7 versions (at least when used with the Omnipod 0.4 dev drivers). Some alerts can already be turned off in the Omnipod preferences (sound for uncertain pump events), but others cannot be turned off or modified. I understand the importance of being notified if delivery is suspended, but the way one is notified and when can dramatically influence perceived alarm fatigue. I am very tired of these alarms and even though AAPS is wonderful in keeping my glucose in range, the addition of these alarms to the app have noticeably reduced my user experience. While surely improving the core features of AAPS is understandably a focus in the app development, I am kindly asking to please not forget about these little things affecting quality of life. Thank you! :)

I am not an App developer so I apologise for only complaining and not actively contributing to a solution myself.

Setup:
Google Pixel 3 XL (Android 11)
AAPS 2.8.2 (algorithm modified)
Omnipod driver (full release)
Glucose data from xDrip+ (Dexcom G6)

rICTx-T1D, 2021-02-04:
I hope the alarm works better than the bonus calculator remember alarm for eating later. The remember sound is anoing too and it was not pushed to WearOS watch.

Please check, that all alarms will send to watch and phone-sound is able to mute and vibrate only. When the alarm is not acknowledged, the alarm will repeate every 5min. So I think its discrete and safe to. Maybe the volume will be begin with single slightly beep and escalate to louder with more tones in sequence of each escalations.

rustymonkey, 2021-11-22:
My son experienced a similar issue. He was in a lecture and AAPS generated the failed ?connection/basal? alarm even though the connection was mostly fine ( this can occur as described at #889 ). When it occurred a second time, he was called out by the teacher (and he just quickly shut down his AAPS phone). It would be a huge usability improvement if these alarms could be better controlled.

The simplest solution might be for the alert to adhere to AAPS alert settings as suggested in #969.

zibra, 2026-08-18:
PR #5086 addresses the failed basal profile update part of this request by adding a Settings > Alerts option to disable its alarm sound while keeping the failure notification visible. It does not implement the broader alarm scheduling or custom-sound requests.