nightscout/AndroidAPS #277: Building in Azure Pipeline

pieteckhart, 2021-01-30:
Hi,

I'm trying to help out my brother in law with building an APK because he was having troubles building it himself.
3 months ago we managed to build one on his PC but now he can't get it to work.

Today I've tried solving his problems by configuring an build pipeline in Azure Devops.
I'm a .Net programmer so I am used to work with Microsoft's products so this would be an easy way to automate this build.
I also have some experience building native Android apps using Xamarin.

Can someone please help me with any of the following?
When I get this working I would gladly share my build pipeline yaml script and instructions so others can also setup a pipeline for themselves.

Thanks!

1. In order to get a build working I needed to upgrade ndkVersion from 21.1.6352462 to 21.3.6528147 because the build agent was based on an image that did not have this specific version. Can this cause problems?

2. I can successfully build (and sign) debug APKs but when I build a release version it fails during the linking process:
```
     com.android.builder.internal.aapt.v2.Aapt2Exception: Android resource linking failed
     /Users/runner/work/1/s/core/src/main/res/layout/maintenance_import_list_item.xml:24: AAPT: error: resource color/importListFileName (aka info.nightscout.androidaps.core:color/importListFileName) not found.
         
     /Users/runner/work/1/s/core/src/main/res/layout/maintenance_import_list_item.xml:34: AAPT: error: resource color/importListFileName (aka info.nightscout.androidaps.core:color/importListFileName) not found.
         
     /Users/runner/work/1/s/core/src/main/res/layout/maintenance_import_list_item.xml:57: AAPT: error: resource color/importListFileName (aka info.nightscout.androidaps.core:color/importListFileName) not found.
         
     /Users/runner/work/1/s/core/src/main/res/layout/maintenance_import_list_item.xml:84: AAPT: error: resource color/importListAdditionalInfo (aka info.nightscout.androidaps.core:color/importListAdditionalInfo) not found.
         
     /Users/runner/work/1/s/core/src/main/res/layout/maintenance_import_list_item.xml:95: AAPT: error: resource color/importListAdditionalInfo (aka info.nightscout.androidaps.core:color/importListAdditionalInfo) not found.
         
     /Users/runner/work/1/s/core/src/main/res/layout/maintenance_import_list_item.xml:116: AAPT: error: resource color/importListAdditionalInfo (aka info.nightscout.androidaps.core:color/importListAdditionalInfo) not found.
         
     /Users/runner/work/1/s/core/src/main/res/layout/maintenance_import_list_item.xml:127: AAPT: error: resource color/importListAdditionalInfo (aka info.nightscout.androidaps.core:color/importListAdditionalInfo) not found.
```
I've checked and found two colors.xml files:
- app\src\main\res\values\colors.xml (seems ok)
- core\src\main\res\values\colors.xml (not containing importListFileName/importListAdditionalInfo)

I've tried to add the seemingly missing colors to the core colors.xml and this results in other build errors:
```
com.android.builder.internal.aapt.v2.Aapt2Exception: Android resource linking failed
     /Users/runner/work/1/s/medtronic/src/main/res/layout/medtronic_fragment.xml:500: AAPT: error: resource string/combo_refresh (aka info.nightscout.androidaps.plugins.pump.medtronic:string/combo_refresh) not found.
         
     /Users/runner/work/1/s/rileylink/build/intermediates/packaged_res/release/layout/rileylink_status.xml:11: AAPT: error: resource style/AppTheme.AppBarOverlay (aka info.nightscout.androidaps.plugins.pump.medtronic:style/AppTheme.AppBarOverlay) not found.
         
     /Users/runner/work/1/s/rileylink/build/intermediates/packaged_res/release/layout/rileylink_status.xml:34: AAPT: error: resource drawable/ic_refresh (aka info.nightscout.androidaps.plugins.pump.medtronic:drawable/ic_refresh) not found.
         
     /Users/runner/work/1/s/medtronic/src/main/res/xml/pref_medtronic.xml:10: AAPT: error: resource string/sixdigitnumber (aka info.nightscout.androidaps.plugins.pump.medtronic:string/sixdigitnumber) not found.
         
     /Users/runner/work/1/s/medtronic/src/main/res/xml/pref_medtronic.xml:21: AAPT: error: resource array/medtronicPumpTypeArray (aka info.nightscout.androidaps.plugins.pump.medtronic:array/medtronicPumpTypeArray) not found.
         
     /Users/runner/work/1/s/medtronic/src/main/res/xml/pref_medtronic.xml:29: AAPT: error: resource array/medtronicPumpFreqArray (aka info.nightscout.androidaps.plugins.pump.medtronic:array/medtronicPumpFreqArray) not found.
         
     /Users/runner/work/1/s/medtronic/src/main/res/xml/pref_medtronic.xml:29: AAPT: error: resource array/medtronicPumpFreqValues (aka info.nightscout.androidaps.plugins.pump.medtronic:array/medtronicPumpFreqValues) not found.
         
     /Users/runner/work/1/s/medtronic/src/main/res/xml/pref_medtronic.xml:61: AAPT: error: resource array/medtronicBolusDelay (aka info.nightscout.androidaps.plugins.pump.medtronic:array/medtronicBolusDelay) not found.
         
     /Users/runner/work/1/s/medtronic/src/main/res/xml/pref_medtronic.xml:69: AAPT: error: resource array/medtronicEncoding (aka info.nightscout.androidaps.plugins.pump.medtronic:array/medtronicEncoding) not found.
         
     /Users/runner/work/1/s/medtronic/src/main/res/xml/pref_medtronic.xml:69: AAPT: error: resource array/medtronicEncodingValues (aka info.nightscout.androidaps.plugins.pump.medtronic:array/medtronicEncodingValues) not found.
         
     /Users/runner/work/1/s/medtronic/src/main/res/xml/pref_medtronic.xml:77: AAPT: error: resource array/medtronicBatteryType (aka info.nightscout.androidaps.plugins.pump.medtronic:array/medtronicBatteryType) not found.
         
     /Users/runner/work/1/s/medtronic/src/main/res/xml/pref_medtronic.xml:77: AAPT: error: resource array/medtronicBatteryTypeValues (aka info.nightscout.androidaps.plugins.pump.medtronic:array/medtronicBatteryTypeValues) not found.
         
     /Users/runner/work/1/s/medtronic/src/main/res/xml/pref_medtronic.xml:85: AAPT: error: resource array/onOff (aka info.nightscout.androidaps.plugins.pump.medtronic:array/onOff) not found.
```
3. Also when running a debug version it looks weird and doesn't seem to work correctly:
[image]

rICTx-T1D, 2021-01-31:
Maybe you build and deploy the WearOS-image on smartphone...

pieteckhart, 2021-01-31:
> Maybe you build and deploy the WearOS-image on smartphone...

I just woke up and realized that it indeed must be the android wear interface :)
Glad to see someone already came to the same conclusion, only six hours before me. I should have gone to bed earlier ;)

I will ty to build the correct apk and will report back if this solves my other problem.

Thanks!

pieteckhart, 2021-01-31:
Ok. I just tried and found out I was building everything correctly (debug mode still) but was copying over the resulting apk when publishing. The following yaml script now builds a debug version:

```
trigger:
- main

pool:
  vmImage: 'macos-latest'

steps:
- task: Gradle@2
  inputs:
    gradleWrapperFile: 'gradlew'
    gradleOptions: '-Xmx3072m'
    publishJUnitResults: false
    testResultsFiles: '**/TEST-*.xml'
    tasks: 'assembleDebug'
    
- task: AndroidSigning@3
  inputs:
    apkFiles: '**/*.apk'
    apksign: true
    apksignerKeystoreFile: 'keystore.jks'
    apksignerKeystorePassword: '********'
    apksignerKeystoreAlias: 'key0'
    apksignerKeyPassword: '*********'
    zipalign: false

- task: CopyFiles@2
  inputs:
    SourceFolder: $(Build.SourcesDirectory)
    contents: '**/*.apk'
    targetFolder: '$(build.artifactStagingDirectory)'
    overWrite: true

- task: PublishBuildArtifacts@1
  inputs:
    pathtoPublish: '$(Build.ArtifactStagingDirectory)/app/build/outputs/apk/full/debug/'
    artifactName: 'apks'
    publishLocation: 'container'
```

pieteckhart, 2021-01-31:
Update: Seems that i needed to specify app:assembleFullRelease instead of 'assembleRelease'. This solved the linking problem. I have now successfully build a release APK :)

I'll ask my brother in law if the resulting APK seems correct. He will test it out on his backup device.
If everything works I will close this issue.

I'm planning to open source this pipeline yaml so others can benefit from it.
Should I fork this repo and add the yaml file? Would @MilosKozak be open to accept a pull request?

guydavies, 2021-01-31:
@MilosKozak is generally open to PRs. This particular doesn't seem to interfere with any other aspects. As long as it is demonstrably robust, it should be good. It sometimes takes a while if he's busy with upcoming release issues. 

You should fork and open a PR from there into dev branch of this repo. I'd be tempted to create a contrib folder since this isn't related directly to the code itself but provides an additional optional approach to deployment.

pieteckhart, 2021-01-31:
@guydavies Thanks for the reply. I'm fairly new to contributing to opensource projects.
I'm reading: https://drewdevault.com/2020/06/06/Add-a-contrib-directory.html

This is what you mean right?

guydavies, 2021-01-31:
Exactly.

pieteckhart, 2021-01-31:
Done: https://github.com/pieteckhart/AndroidAPS/tree/master/contrib/azure-pipelines
I'll make a PR when it's polished enough for my taste ;)

closed 2021-01-31 by pieteckhart