nightscout/AndroidAPS #283: Incorrect treatments uploaded into nightscout

dmelliott20, 2021-02-05:
Pump: Insight.
Version 2.8.2

Ive noticed spurious entry's in nightscout of incorrect basal rates being uploaded into nightscout.
I have removed all connections to nightscout from xdrip etc to make sure that android aps is the only one uploading.

see duplicate entries below. the 3.72 rate is way above any safety limits set on the loop. the basal is not being enacted on the pump. its just an incorrect upload of the data. 

{
    "_id": "601150becce8360004cf5b92",
    "eventType": "Temp Basal",
    "duration": 9,
    "absolute": 3.72,
    "rate": 3.72,
    "pumpId": 1595582019158,
    "created_at": "2021-01-27T11:28:39Z",
    "enteredBy": "openaps://AndroidAPS",
    "originalExtendedAmount": 0.54,
    "NSCLIENT_ID": "1611747518967",
    "carbs": null,
    "insulin": null
  },
  {
    "_id": "60114e84cce8360004cf5b8b",
    "eventType": "Temp Basal",
    "duration": 30,
    "absolute": 1.2000000000000002,
    "rate": 1.2000000000000002,
    "pumpId": 1595582019158,
    "created_at": "2021-01-27T11:28:39Z",
    "enteredBy": "openaps://AndroidAPS",
    "originalExtendedAmount": 0.54,
    "NSCLIENT_ID": "1611746919601",
    "carbs": null,
    "insulin": null
  },

Philoul, 2021-10-04:
See below screenshots of AAPS compared to NS, everything is Ok for me:
On the 2 screenshots on the left we can see standard basal rates (with TBR emulation enabled)
=> We have 2 extended bolus during this period

Then I disabled TBR emulation and set manually an extended bolus for about 8 min

And I enabled again in Insight settings TBR emulation to be able to compare the curves (2 screenshots on the left)

[image]

dmelliott20, 2021-10-05:
It looks fine via screen shots on mine too it’s the back end treatment data that’s the issue.

MilosKozak, 2021-10-05:
what version did you test? @dmelliott20

dmelliott20, 2021-10-05:
Hi Milos

Im on 2.82

see the treatments below. - they appear to be duplicated and are incorrect (they are way outside my safety nets set within android aps)

on the front end both in androidaps and nightscout the basal rates look correct and are reported correctly.

[image]

dmelliott20, 2021-10-05:
[image]

MilosKozak, 2021-10-05:
@dmelliott20 we are working and adding changes to dev branch 2.8.2.5-dev currently
did you test this version?

dmelliott20, 2021-10-05:
Ah no I haven’t. I probably won’t be able to test this week.

closed 2021-12-07 by MilosKozak