nightscout/AndroidAPS #287: 2.8.2.1-dev: No bg readings from xDrip+

swissalpine, 2021-02-06:
Since yesterday I get no readings with actual xDrip+ dev
```
15:52:00.235 [main] D/DATASERVICE: [DataReceiver.onReceive():36]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => -9.989244912977028E-6; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2101292241 e89e23d-2021.01.29; com.eveningoutpost.dexdrip.Extras.Raw => 0.0; com.eveningoutpost.dexdrip.Extras.Time => 1612623116910; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => -1; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => -1; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 186.0; com.eveningoutpost.dexdrip.Extras.SourceDesc => G6 Native / G5 Native; com.eveningoutpost.dexdrip.Extras.SourceInfo => G5 Native; }Bundle
15:52:00.267 [pool-14-thread-1] D/BGSOURCE: [XdripPlugin$XdripWorker.doWork():60]: Received xDrip data: Data {data : , collection : null, }
```
[image]

rkresha, 2021-02-09:
same here. Had to revert to my build from 1/30

swissalpine, 2021-02-09:
Hi Adrian,
first impression: oh God, i lost all readings (display: 0.00, disappeared graph).
But then I got xDrip+ readings and the grah is rebuilt reading by reading.
If the lost of readings history is intended, than the fix is o.k.
Once more: THANK YOU!

AdrianLxM, 2021-02-09:
@swissalpine the loss of history is a side effect of switching to a modern database system and unrelated. (parallel development).
Thanks for testing.

closed 2021-02-09 by AdrianLxM