nightscout/AndroidAPS #289: Incomplete Export Settings imported transferred for v2.8.2 from old format

JustinSouthcombe, 2021-02-07:
When upgrading (today 12:59 GMT) to v2.8.2 (new format) I am finding the Export Settings "Preferences" file from v2.7.0 (old format) is not being completely transferred when I import it. Some key values missing/old/different. 
[AndroidAPS[1].log](https://github.com/nightscout/AndroidAPS/files/5939390/AndroidAPS.1.log)
 Local profile is completely different for example (additional profiles absent) and the pump settings (e.g max bolus) different as well. I have attached the logs below. 
Q. Do I need to complete the new Objectives before the Local Profile will etc will update correctly? 

[AndroidAPS_LOG_1612713608289.log[1].zip](https://github.com/nightscout/AndroidAPS/files/5939415/AndroidAPS_LOG_1612713608289.log.1.zip)

MilosKozak, 2021-02-07:
old format is not supported anymore, but it can work if you are lucky
it's not related to objectives at all

closed 2021-02-07 by MilosKozak

JustinSouthcombe, 2021-02-07:
Should I sync profiles with NS and then download from there to v2.8.2 ?