nightscout/AndroidAPS #293: feature request 2.8.2 amend "record Bolus only" to calculator

cheers2, 2021-02-08:
could you please amend the "record only Bolus" tick box of the Insulin Input screen also to the calculator.

Helps to calculate external Bolus and record it at the same time w/ having to switch to carb and insulin input screens after calculation. Avoids mistakes. Many thanks.

MilosKozak, 2021-02-11:
what's the usecase?

cheers2, 2021-02-12:
Thanks for reply.

I have fairly high insulin doses per meal. In average 15IE with a max to 20IE. I also have problems with my canula sites, so the doc recommended to alternate injection sites. Therefore I usually use external Bolus for meals with high amounts of insulin, basically 3 times a day. Often I also split these boluses to two injection sites.

Process then is: 
1 Calculate Bolus in the calculator by carbs input
2 receive calulated insulin dose
3 cancel calc
4 go to carbs input, input carbs again
5 go to insulin input, input calculated insulin dose (happens that I have already forgotten the exact dose)
6 inject insulin

With the tick box I could skip the 3 steps 3, 4, 5 in between and also avoid mistakes.

Position of the tick box is not important. It could also be below the section of the exact calculation, which can be switched on and off, to not disturb the main calculator screen.

MilosKozak, 2021-02-12:
i understand your stand but it's still rare case. wizard is the most used dialog and keeping it small and simple saves time to everyone. adding more controls for rare cases goes directly against it.
So sorry but no for now

closed 2021-02-12 by MilosKozak

cheers2, 2021-02-12:
Ok understood,

you are right, it is a rare case, but maybe you can think of anything else in the future, but thanks for the reply and explanation.