nightscout/AndroidAPS #295: [2.8.2] Insulinampullenwechsel - Accu-Check Insight Pumpe / Insulin reservior changes - Accu-Check Insight Pump

TeufelHH, 2021-02-08:
Wenn ich bei der Accu-Check Insight Pumpe die Insulinampulle wechsle wird das Alter von dem Katheter ebenfalls zurück gesetzt. Ich starte den Ampullenwechsel nicht über AAPS sondern direkt über die Pumpe.

- Pumpe -> Ampulle und Infusionsset -> Ampulle wechseln. 
- Danach Schlauch füllen
- Insulinabgabe starten

Danach wird in AAPS unter "Behandlungen -> Careportal" folgendes geloggt:

- Insulinreservoir-Wechsel (korrekt)
- Schlauch gewechselt (korrekt)
- Pumpenkatheter-Wechsel (falsch)

Meine Pumpe ist die Accu-Check Insight v3 (ist eine Leihpumpe) mit meiner eigenen Accu-Check Insight v1 hatte ich das Problem nicht, da wurde nur der Reservior & Schlauchwechsel geloggt.

Ich weiß nun nicht ob es an der Pumpe liegt, oder an AAPS 2.8.2 mit meiner AC Insight V1 und AAPS 2.7.0 gab es dieses Problem nicht.

Google Translate:
If I change the insulin cartridge on the Accu-Check Insight pump, the age of the catheter is also reset. I don't start the ampoule change via AAPS but directly via the pump.

- Pump -> ampoule and infusion set -> change ampoule.
- Then fill the hose
- Start insulin delivery

Then the following is logged in AAPS under "Treatments -> Careportal":

- Change of insulin reservoir (correct)
- hose changed (correct)
- Pump catheter change (wrong)

My pump is the Accu-Check Insight v3 (is a rental pump) with my own Accu-Check Insight v1 I did not have the problem, only the reserve and hose change were logged.

I don't know if it's the pump, or AAPS 2.8.2 with my AC Insight V1 and AAPS 2.7.0 didn't have this problem.

reijoahola-nightscout, 2021-02-09:
This must be some difference in Insight v3 because I do not have this problem with Insight v2. I expect that when you say "age of catheter is also reset" it actually means "age of canula (site) is reset". You can temporarily avoid the problem by turning off logging of reservoir changes (AAPS Insight settings) and by manually giving a comnand "Actions/Insulin Cartridge Change" in AAPS.

TeufelHH, 2021-02-09:
Unfortunately, under "Actions" in AAPS I don't have the item "Change insulin cartridges". I can only trigger the whole thing with the pump. Also in Docs recently it says that this point no longer exists.
[image]

reijoahola-nightscout, 2021-02-09:
Insulin Cartridge Change command is under that Katheterwechsel button.

TeufelHH, 2021-02-09:
I haven't seen that yet. In the past there was a single button for this, I think I can remember. Thanks for the information.

reijoahola-nightscout, 2021-02-09:
[image]

Philoul, 2021-02-12:
I confirm I have issue with my Insight v3 pump.
It worked correctly when I started to loop in the end of 2019, with a firmware v2 and an older version of AAPS...
I can try to connect my old pump on my dev phone to see if it works or not... (I thought it was associated to an AAPS update, but maybe it's linked to insight firmware modification...)...
@TebbeUbben do you think you can fix this issue?

khind, 2021-02-12:
Same problem with the Dana RS

closed 2021-10-04 by MilosKozak