nightscout/AndroidAPS #300: Bolus percentage improvements in bolus wizard, and quickwizard

szantos, 2021-02-10:
[image]

1) The units from "Deliver this part of bolus wizard result" should be considered before the Corr
   (it would be easier to manage the correction)

2) "Deliver this part of bolus wizard result" should have an own row within the details with the corresponding units on the right (maybe with a fine line above - meaning, that all above has been considered)

3) optionally with a checkbox

4) but more important with an input field for overriding the standard value
   example use cases:
   - if there is more active insulin on meal start, then you have to prebolus less
   - if you know, you'll eat faster acting carbs, you have to prebolus more, maybe 100%

5) in the row with the "Result:" 60% could be on the left, and Result: 1,30U on the right
   (stating, that 60% has already been considered)

6) Addition of Carb duration field under Carb

7) Moving Corr input field to the very bottom, or under Carb time
   (that the 3 Carb input fields are beside each other)

8) Ability to consider custom bolus percentage for each quickwizard button (either before creating the button for the button itself, or after pressing the quickwizard button on the approval window)

Thanks & feel free to discuss :)

-------------------------------

This is how we do it right now:
- input all carbs to bolus wizard to get 100% bolus (with every checkbox except TT checked)
- cancel
- calculate (in the head) the exact pre-bolus considering a custom %
- delivering pre-bolus with the "Insulin" button
- waiting for delivery
- entering extended Carbs with the "Carbs" button
All in all: you have to use 3 buttons (Bolus Wiz, Insulin, Carbs); calculate in head; wait for insulin delivery

MilosKozak, 2021-02-12:
wizard is the most used dialog and keeping it small and simple saves time to everyone. adding more controls for rare cases goes directly against it.
So sorry but no for now

szantos, 2021-02-12:
@MilosKozak 
Thanks for considering.
1,2,5,8,9,10 are not about adding anything new. They would help the understanding, or would put some things in the right place (considering corr after %; no superbolus if %).

Philoul, 2021-02-12:
@szantos FYI You can take a look on this proposal close to some of yours and associated comments... https://github.com/MilosKozak/AndroidAPS/issues/2462

Philoul, 2021-03-06:
@MilosKozak do you think we can have 2 "level" of layout for calculator (short click for a simple one, long click with more fields as proposed here)
- I think we could simplify current layout for a short click if we hide BG field if bg value is available, and the row with profile and superbolus
- For a long click (advanced settings) all requested fields here : carb duration, super bolus duration, override default % of bolus...

Side question, how we calculate a bolus for extended carbs without safety risks if a long duration is selected ?

szantos, 2021-03-10:
@Philoul 
Your side question is actually a very good one, as extended carbs are not covered by the bolus calculator, and it is the right way.
I'll try to figure out a new concept.

Andries-Smit, 2022-03-07:
@szantos point 8 was implemented in PR #1317