nightscout/AndroidAPS #302: NightscoutClient crashes on actual dev

swissalpine, 2021-02-10:
AndroidAPS crashes when connecting to heroku
- Upload disabled but NS Profiles active

`2021-02-10 21:59:52.899 3759-3759/? E/jamorham nsreceiver: Unknown action! info.nightscout.client.NEW_PROFILE
2021-02-10 21:59:52.901 16773-16773/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: info.nightscout.androidaps, PID: 16773
    java.lang.IllegalStateException: Data cannot occupy more than 10240 bytes when serialized
        at androidx.work.Data.toByteArrayInternal(Data.java:417)
        at androidx.work.Data$Builder.build(Data.java:846)
        at info.nightscout.androidaps.receivers.DataReceiver.onReceive(DataReceiver.kt:66)
        at androidx.localbroadcastmanager.content.LocalBroadcastManager.executePendingBroadcasts(LocalBroadcastManager.java:313)
        at androidx.localbroadcastmanager.content.LocalBroadcastManager$1.handleMessage(LocalBroadcastManager.java:121)
        at android.os.Handler.dispatchMessage(Handler.java:107)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7386)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:980)
        `
Version:
2.8.2.1-dev
c0a90d010-2021.02.09-21.07

AdrianLxM, 2021-02-10:
@MilosKozak it seems that profiles can be too large to directly send it as Data to a Worker.
Shouldn't we just use the `bundleStore` for all Workers started by `DataReceiver`? (I can do that, if you want)

MilosKozak, 2021-02-11:
fixed by https://github.com/nightscout/AndroidAPS/commit/7acaf8a7e67fab5df8139f5be52c55080f3eae49

closed 2021-02-11 by MilosKozak