nightscout/AndroidAPS #311: OmniPod fault 049 INSULIN_DELIVERY_COMMAND_ERROR when temp basal is automatically set

kalleelmer, 2021-02-11:
My pod suddenly started screaming and i got a 049 error. The phone was unused in my pocket at the time.

Looking as the logs, the error appears to happen when the app is trying to set a temporary basal in response to a new glucose reading. It begins at 2020-02-10 13:16:29 when a new glucose reading is received and results in a PodFaultException at 13:16:36.

I use an EmaLink. Glucose readings are received from a Libre 2 using the patched LibreLink app and XDrip.

AndroidAPS 2.8.2
Build: 194549adc-2021.01.23-19:55
Flavor: fullrelease
Nightscout version: 14.0.2

[aaps-pod-error-49-excerpt.log](https://github.com/nightscout/AndroidAPS/files/5964313/aaps-pod-error-49-excerpt.log)
[aaps-logs.zip](https://github.com/nightscout/AndroidAPS/files/5964315/aaps-logs.zip)

lkiversen, 2021-02-23:
I have the same issue 4 times in 12 days.
I use omnipod and dexcom g6

samspycher, 2021-10-05:
@kalleelmer @lkiversen are you still seeing this issue?

kalleelmer, 2021-10-05:
@samspycher I haven't seen this since I reported it. Hopefully it is gone forever.

nlof, 2022-08-03:
I just had the same fault (049) for the first time in a year of using AAPS. Should I send logs?

nlof, 2022-08-07:
It starts to be bizzare. 
Second pod fault command error in 4 days. This is after moving to a new phone and upgrading from v. 2.8.2.1 to 3.1.0.3. 

Pods are from a batch which has expired about 30 days ago. I wonder whether this could be an explanation. Again, the new pod was only a few hours old. In between I had a normally working pod from the same batch.

If nobody comments on this I'll open a new ticket...

[image]

Milanakryskovskaya, 2024-04-23:
Error 049 is repeated regularly. Over the past month, 5 Omnipods have broken. The shelf life is good. Two Omnipods have broken down for the last two days in a row. AAPS 3.2.0.3.  
[Uploading AndroidAPS._2024-04-22_06-16-16_.0.zip…]()
 I don't know what to do ((

Milanakryskovskaya, 2024-04-23:
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/15082669/AndroidAPS.log)
[AndroidAPS._2024-04-22_06-16-16_.0.zip](https://github.com/nightscout/AndroidAPS/files/15082673/AndroidAPS._2024-04-22_06-16-16_.0.zip)
Krutko5638@gmail.com

samspycher, 2024-04-30:
HI @Milanakryskovskaya 

Unfortunately it has been a long time since I have worked on this code, but I did take a quick look:

The error seems to be ocurring during delivery of temporary basals. The following log statements show one example each of successful delivery and failed delivery as per your log files:

```
FAILED DELIVERY OF TEMP BASAL (24th April 2024, 05:48):
# Request to set temp basal with SetInsulinScheduleCommand
05:48:20.395 [Thread-1249] D/PUMPBTCOMM: [OmnipodRileyLinkCommunicationManager.exchangeMessages():116]: Exchanging OmnipodMessage: responseClass=[StatusResponse, AapsPodStateManager{podState=PodState{address=520749791, lot=72446, tid=3210371, piVersion=2.10.0, pmVersion=2.10.0, packetNumber=19, messageNumber=0, lastSuccessfulCommunication=2024-04-22T05:18:30.790+03:00, lastFailedCommunication=2024-04-22T05:12:26.246+03:00, lastUpdatedFromResponse=2024-04-22T05:18:30.790+03:00, timeZone=Europe/Minsk, activatedAt=2024-04-21T23:37:04.002+03:00, timeActive=PT20520S, faultEventCode=null, reservoirLevel=null, totalTicksDelivered=315, suspended=false, nonceState=NonceState{table=[333211736, 1401021905, -920135861, -1507717511, -1989628444, 722167411, 218647397, 477991783, -1143347759, 341010232, -432244307, 1678827731, 1218914998, 351384896, 495017846, 1553110352, -657701264, 1561499653, 0, 0, 0], index=10}, activationProgress=COMPLETED, podProgressStatus=ABOVE_FIFTY_UNITS, lastDeliveryStatus=TEMP_BASAL_RUNNING, activeAlerts=AlertSet{alertSlots=[]}, basalSchedule=BasalSchedule{entries=[BasalScheduleEntry{rate=1.55, startTime=0s}]}, lastBolusStartTime=2024-04-21T23:41:29.774+03:00, lastBolusAmount=1.0, lastBolusDuration=PT40.689S, lastBolusCertain=true, tempBasalAmount=1.95, tempBasalStartTime=2024-04-22T05:48:21.874+03:00, tempBasalDuration=PT1800S, tempBasalCertain=false, expirationAlertTimeBeforeShutdown=null, lowReservoirAlertUnits=null, configuredAlerts={SLOT2=SHUTDOWN_IMMINENT_ALARM}}}, OmnipodMessage{address=520749791, messageBlocks=[SetInsulinScheduleCommand{schedule=TempBasalDeliverySchedule{secondsRemaining=1800, firstSegmentPulses=19, basalTable=BasalDeliveryTable{entries=[BasalTableEntry{segments=1, pulses=19, alternateSegmentPulse=false}]}}, nonce=1218914998}, TempBasalExtraCommand{acknowledgementBeep=false, completionBeep=false, programReminderInterval=PT0S, remainingPulses=19.5, delayUntilNextPulse=92.30769230769232, rateEntries=[RateEntry{totalPulses=19.5, delayBetweenPulsesInSeconds=92.30769230769232}]}], encoded=1F0A02DF00201A0E48A72AB601009F01384000130013160E000000C3008CD9B100C3008CD9B182E0, sequenceNumber=0}, null, null, true], podStateManager={}, message={}, addressOverride={}, ackAddressOverride={}, automaticallyResyncNonce={}
# Receiving response status INSULIN_DELIVERY_COMMAND_ERROR
05:48:27.006 [Thread-1249] D/PUMPBTCOMM: [OmnipodRileyLinkCommunicationManager.exchangeMessages():134]: Received response from the Pod [responseMessageBlock=[PodInfoResponse{subType=DETAILED_STATUS, podInfo=PodInfoDetailedStatus{podProgressStatus=FAULT_EVENT_OCCURRED, deliveryStatus=SUSPENDED, bolusNotDelivered=0.0, podMessageCounter=14, ticksDelivered=338, insulinDelivered=16.900000000000002, faultEventCode=Pod fault (49): INSULIN_DELIVERY_COMMAND_ERROR, faultEventTime=PT22320S, reservoirLevel=null, timeActive=PT22320S, unacknowledgedAlerts=AlertSet{alertSlots=[]}, faultAccessingTables=false, errorEventInfo=ErrorEventInfo{insulinStateTableCorruption=false, internalVariable=0, immediateBolusInProgress=false, podProgressStatus=ABOVE_FIFTY_UNITS}, receiverLowGain=0, radioRSSI=30, previousPodProgressStatus=ABOVE_FIFTY_UNITS, unknownValue=00 8C}}]]
# Exception stack trace
05:48:27.021 [Thread-1249] E/PUMP: [AapsOmnipodErosManager.handleException():915]: Caught OmnipodException[certainFailure=true] from OmnipodManager
info.nightscout.androidaps.plugins.pump.omnipod.eros.driver.exception.PodFaultException: Pod fault (49): INSULIN_DELIVERY_COMMAND_ERROR
	at info.nightscout.androidaps.plugins.pump.omnipod.eros.rileylink.manager.OmnipodRileyLinkCommunicationManager.exchangeMessages(OmnipodRileyLinkCommunicationManager.java:170)
	at info.nightscout.androidaps.plugins.pump.omnipod.eros.rileylink.manager.OmnipodRileyLinkCommunicationManager.exchangeMessages(OmnipodRileyLinkCommunicationManager.java:108)
	at info.nightscout.androidaps.plugins.pump.omnipod.eros.rileylink.manager.OmnipodRileyLinkCommunicationManager.exchangeMessages(OmnipodRileyLinkCommunicationManager.java:104)
	at info.nightscout.androidaps.plugins.pump.omnipod.eros.driver.communication.action.SetTempBasalAction.execute(SetTempBasalAction.java:45)
	at info.nightscout.androidaps.plugins.pump.omnipod.eros.driver.communication.action.SetTempBasalAction.execute(SetTempBasalAction.java:16)
	at info.nightscout.androidaps.plugins.pump.omnipod.eros.rileylink.manager.OmnipodRileyLinkCommunicationManager.executeAction(OmnipodRileyLinkCommunicationManager.java:100)
	at info.nightscout.androidaps.plugins.pump.omnipod.eros.driver.manager.OmnipodManager.lambda$setTemporaryBasal$7$info-nightscout-androidaps-plugins-pump-omnipod-eros-driver-manager-OmnipodManager(OmnipodManager.java:259)
	at info.nightscout.androidaps.plugins.pump.omnipod.eros.driver.manager.OmnipodManager$$ExternalSyntheticLambda3.get(Unknown Source:10)
	at info.nightscout.androidaps.plugins.pump.omnipod.eros.driver.manager.OmnipodManager.executeAndVerify(OmnipodManager.java:551)
[... cut ...] 4.0, delayBetweenPulsesInSeconds=75.0}]}], encoded=1F0A02DF38201A0EF04179AD0100A901384000180018160E000000F0007270E000F0007270E00253, sequenceNumber=14}, null, null, true], podStateManager={}, message={}, addressOverride={}, ackAddressOverride={}, automaticallyResyncNonce={}
```

However, this error was fixed, and I don't see any time change in your logs, so your error 049 may not be time-related. But just to verify: do you know of anything time-change related that may be happening to your phone?