nightscout/AndroidAPS #334: DanaRS "Wrong Pump Password" despite correct password... limited functionality.

BlockSpaceVictor, 2021-02-14:
Hi, maybe this is a known issue but I didn't see anyone else post about it. I built the latest APK 2.8.2 and installed it successfully on the phone. It tries to connect to the pump (DanaRS), and shows BG value, but gives me error message "Wrong Pump Password", which leads to limited functionality. For example, "Insulin", "Carbs", and "Calculator" functions are not showing up at all (see image). The only thing showing up "CGM". I'm using modified Dexcom app. 

Checking the password in the pump (default 0000) and the password settings in AAPS, they match. Yet, no cigar. Help. 

[image]

MilosKozak, 2021-02-14:
https://androidaps.readthedocs.io/en/latest/Configuration/DanaRS-Insulin-Pump.html?highlight=DanaRS

AdrianLxM, 2021-02-14:
You can use the default password to set a proper one that can be used for communication.
Depending on version on the firmware `0000` will not work directly.

MilosKozak, 2021-02-14:
you probably upgraded from very old version.
check release notes of all version between

BlockSpaceVictor, 2021-02-14:
I upgraded from 2.6.1, never had any issues. 

I'm not sure how to understand the posted link... even if I change pump password in both pump and AAPS settings, it does not work. 

I checked and the firmware of the DanaRS is version 1. Maybe that has something to do with it?

MilosKozak, 2021-02-14:
2.6.1 didn't verify RS password
just change it on pump (to prove you know it) and put to AAPS

BlockSpaceVictor, 2021-02-14:
Yeah, I changed the password several times, from 0000, to 1234, to other numbers... but it never works. Of course, I also change it in the Dana settings on AAPS. But no luck.

MilosKozak, 2021-02-14:
in order to change RS pass you need to enter old pass to pump first.
So i'd assume you don't know it

BlockSpaceVictor, 2021-02-14:
ohhhhh. Ok now it works. The issue was that I thought I was changing the password in the pump, when in fact I just failed to correctly input the old password, because Dana gives no error message if you put in the wrong one. So it was just me being stupid. Thanks for the quick replies! :)

closed 2021-02-14 by BlockSpaceVictor