nightscout/AndroidAPS #335: Tidepool

paddingtonil, 2021-02-14:
I would like to request feature support for tidepool. Health care professionals are using tidepool as main source for bg and insulin data. Unfortunately nightscout is not widely accepted by many physicians hence uploading directly from aaps to tidepool is important. Currently xdrip allows uploading, but very limited support.

andyrozman, 2021-02-14:
Problem is that Tidepool doesn't allow to upload everything (profile is big problem), and until they extend their APIs there is nothing for us to do. 
Tidepool is currently working on their Loop solution, so they don't have time to do anything else.

JohnKitching, 2021-02-15:
If TidePool "want" DIY data and Nightscout is used by the DIY community shouldn't TidePool read from Nightscout?

guydavies, 2021-02-15:
I’m not sure that Tidepool “wants” this data.  It is the user community that is pushing for it.  Tidepool has been reasonably successful in getting some clinics to start using their platform as the defacto standard platform.  It has a reasonably wide range of BG measurement devices supported and a few pumps from which it can import data.  If your doctor wants your data in Tidepool then I don’t suppose that the patients really care about the mechanism whereby that data gets into Tidepool, but either way, Tidepool needs to make some modifications to support the profile model supported in AAPS and Nightscout before data can be reliably exchanged.

andyrozman, 2021-02-15:
@JohnKitching 
Tidepool is sort-of "replacement" for Nightscout, so I don't think they would want to read from it. 

Theoretically we could upload some data to Tidepool even now, but we can't upload all the data, so we couldn't run AAPS directly from it. 

So this ticket at the moment is too wide. @paddingtonil What did you want to do with Tidepool and AAPS? Just have data there (just uploading) so that your doctor can see it, or use data from there in AAPS (that would require also downloading, of in some case cgms data, treatment data, profile data (optional)).

lee-b, 2021-02-24:
I think the most important thing, by far, is to get carbs and doses into tidepool, since there are other ways to get BGs in there.  The combination of the two in one platform that has good reporting, no setup, and can easily be shown to medical professionals is not to be underestimated.

I've no real interest in uploading a profile, for example.  Nice to have, but carbs and doses are what gets laboriously entered into AAPS, and what is valuable to export to a common visualisation/analysis tool.

rICTx-T1D, 2021-04-15:
Maybe it's possible to create a separate module in AAPS (simular to openHumansUploader) to upload Carbs and Isulin doses to Tidepool. All other information,  e.g. Notes, Profiles, Exercises,... can be transferred later, when API will support this.

MilosKozak, 2021-04-16:
without profile the insulin dosage is confusing because TDD will not match the reality
Tidepool plugin already exists but it's off because we cannot upload data correctly

yhonatanb20, 2021-11-01:
Please enable the tidepool plugin again, it's very important and I'll try to explain why:
1. In order to use android aps, you have to create a nightscout site, which is a bit complicated and confusing, or otherwise pay for it, which kind of ruins the whole DIY idea. Tidepool, however, is much more user friendly, is completely free and is by far more supported by clinitians.
2. I can hardly see what is the problem about the profile. If tidepool can receive all data including boluses, basal and temp basals from omnipod, why shouldn't it be able to do the same with android aps? It also supports xdrip data uploading which may make the whole business easier.
Please, if possible, add at least some support for tidepool. I'm sure that many android aps users (including me) will be very thankful for this.

janvitos, 2022-01-06:
Hi! I was wondering if there's any updates on this ?

We also normally use Tidepool to share our diabetic child's data with his endocrinologist, and having the basal uploaded from AndroidAPS to Tidepool is a must. Since we switched from regular Omnipod to AndroidAPS, we no longer have the basal in Tidepool, so we'll have to send his endo the reports generated from Nightscout for now. Hopefully, this will do for the time being.

Thanks!

fglta, 2022-04-24:
> Problem is that Tidepool doesn't allow to upload everything (profile is big problem), and until they extend their APIs there is nothing for us to do. Tidepool is currently working on their Loop solution, so they don't have time to do anything else.

> Tidepool plugin already exists but it's off because we cannot upload data correctly

@andyrozman @MilosKozak apologies for the naive question but I could use a pointer. Could you elaborate on what is it that is missing? I was casually looking into this, and came across https://tidepool.atlassian.net/browse/BACK-2019 and https://tidepool.atlassian.net/browse/BACK-2020 which seem to suggest that Tidepool's platform has gained some support for looping profile fields (geared towards Tidepool Loop but potentialy generic enough?). At minimum, this is a self-contained change that can help guide the addition of additional fields that AAPS may require. I would love to hear from from you and some guidance on how to proceed. Thank you!

lee-b, 2022-06-04:
> without profile the insulin dosage is confusing because TDD will not match the reality Tidepool plugin already exists but it's off because we cannot upload data correctly

What do you mean by "cannot upload data correctly"?  Is there a tidepool API limitation?  An androidAPS tidepool plugin limitation?  Or are you continuing the thought about needing profile upload to give the existing upload code additional context?

If the latter, are you saying that the existing code needs reworked to upload values relative to a profile baseline -- to upload dose corrections above/below the profile's default basal rate, for example?

What _exactly_ is the problem?

andyrozman, 2022-06-04:
Tidepool profile is much simpler than that of NS. It doesn't support all the features, so we loose a lot of information. I haven't worked on Tidepool support, so I can't give you exact info... 
I heard that they extended profile in last few months, so maybe it would be enough... I think that Xdrip already extended their Tidepool code... Maybe someone will do this here. Problem is that we have limited number of developers, and even thoose we have have their own life outside AAPS, so we are limited with what we can do.

koelewij, 2023-01-02:
Is it clear what the status of the current Tidepool plugin is?
Since AAPS V3 can work with an offline profile, I don't see the need to be able to fully upload profiles to Tidepool or vice versa, but just being able to upload what is available. 
The fact that Tidepool is widely accepted, does not require setting up a web app and is free, makes it an interesting alternative for Nightscout. I have shared my NS link with my endo, but they don't feel so comfortable with it as Nightscout is. 
If we could only upload BG, Basals and treatments, it would already be a great 2nd option to report to your medical team. 
However, i have no clue, how much effort is still needed to get this plugin to work.

I am willing to spend effort in testing this plugin when that would be required.

closed 2023-08-17 by MilosKozak