nightscout/AndroidAPS #341: Feature request : add the possibility to inject later (or set a reminder alert)

robertrub, 2021-02-16:
I'm a _slow CHO digesting_ person. Which results in low BGs if I inject insulin at the start of meals. I note carbs input in AAPS and start the meal and I use the calculator towards the end of the meal to inject insulin.

It would be useful if in the calculator I could set either a _wait x minutes before injection_ option or, if it is not easy to add this option, be able to set an alert for a _reminder in x minutes_ to inject insulin

rICTx-T1D, 2021-02-17:
Hi, I have sometimes the same problem. 
It would be good with 2 options:
* remember in X minutes 
* remember when BG is over x mmol/l or mgd/dl

muthijammy, 2021-02-20:
Yes, would love it. Of course, existing automation can be improved to do this bolus? Announcing carbs and forgot to bolus later happened too often

MilosKozak, 2021-02-20:
wizard is designed to be used at time of injecting insulin. use wizard at the end of meal and set negative carbs time

MilosKozak, 2021-02-20:
automation cannot do a bolus. it could be dangerous

MilosKozak, 2021-02-20:
wizard dialog is most used dialog and i we must keep it simple. 
options you are asking would lead to complicated UI

closed 2021-02-20 by MilosKozak