nightscout/AndroidAPS #345: Basal incorrectly recognised as 0 with latest Dev on DanaRSv3

guydavies, 2021-02-18:
I just tried the latest dev build 2b3fcc60f and, while it fixed my previous issue, it now raises a new issue.

When started, it appears that the app correctly connects to the pump and it *appears* to pull down the required data from the pump, except that the quantity of Insulin and the battery level are not recorded.  As soon as the app tries to run the OpenAPS SMB algorithm, it complains that the base Basal is set out of bounds (0) and has been set to 0.01.  This is not the case as my profile currently has a basal of 0.20u/h.

This occurred around 13:14 CET.  At 13:16 CET, I reverted to my previous dev build based on 32df0b949 and that recovered immediately. Logs from my phone are attached.

[AndroidAPS_LOG_1613650784339.log.zip](https://github.com/nightscout/AndroidAPS/files/6002676/AndroidAPS_LOG_1613650784339.log.zip)

MilosKozak, 2021-02-20:
should be fixed now

closed 2021-02-20 by MilosKozak

guydavies, 2021-02-21:
Thanks Milos. It appears to be OK on first glance. I'll run it this
afternoon and get back to you to confirm.