nightscout/AndroidAPS #347: Enhanced automation

Air69, 2021-02-19:
Is it possible to consider adding to the automation to disable (smb) super micro bolusese when the short average change exceeds a certain value, so as to avoid the noise data of G6 at certain times.

TranceCake, 2021-03-30:
Another great addition would be to detect what the value of a temp target is (so you can differentiate low temps from high temps)

TranceCake, 2021-05-15:
I have spoken with someone that did it the other way around, he only has SMB enabled with low temp targets and when he goes high he uses automation to set a low temp target. This way you can also cancels it if you see a delta you don't like, just cancel the temp target. Would be even easier if automation could detect the value of the temp target

Philoul, 2021-10-09:
> Another great addition would be to detect what the value of a temp target is (so you can differentiate low temps from high temps)

This feature is now included in dev

closed 2021-10-09 by MilosKozak