nightscout/AndroidAPS #349: UI feedback Feb 2021

jamespaulley, 2021-02-19:
Some of these ideas are continuations from the following thread.
https://github.com/nightscout/AndroidAPS/issues/67

Below is a bullet list summarising the suggestions. These are all illustrated in more detail in the PDF below.
https://drive.google.com/file/d/1QkuXciiruTB-_1ulfIFZ__8h7SVmSnYw/view?usp=sharing

1. **Internal notification design**. Currently quite messy. More padding needed.
2. **Profile and target buttons on home screen.** More padding needed. Can we swap around the profile button's short and long press functionality?
3. **Bluetooth connection status text.** Currently this obscures the Profile and Target buttons. Can this be placed at the bottom of the home screen instead?
4. **pop-up interface issues.** There are UI issues with the pop-up interfaces for Profile switch and temporary target:
 a. Interface slides around when keypad appears. Let's snap the whole UI to the top.
 b. The "-/+" buttons need more padding and spacing.
 c. Remove background transparency. You can't click on it to close the UI, so it is not needed.
5. **Temporary target icons**
 a. The main temporary target icon is too confusing. I suggest the archery icon as shown in the PDF. I think it should be desaturated also.
 b. The button icons in the temporary target UI are too confusing. I suggest the icons shown in the PDF, again desaturated. The colour theory across AndroidAPS is too confusing and colours are being over-used without a coherent colour theory.

I'd love to know what you think of these suggestions. Thanks.

hartlijn, 2021-05-07:
Good ideas, imo

jamespaulley, 2023-01-12:
Hi, I've updated this list for 2023, removing completed or less-important items. Most of these ideas are illustrated in the PDF found here.
https://drive.google.com/file/d/1QkuXciiruTB-_1ulfIFZ__8h7SVmSnYw/view?usp=sharing

### Bolus Wizard UI

Carbs +/- buttons currently advance by 1g. Can we have the option to specify the jump interval. Personally I would like 10g.

### Android Toast notification blocks keypad

These Toast notifications block the keypad which is super frustrating.
https://imgur.com/a/A18OoG5

### Profile switch UI
Can we have the option to specify the intervals for the +/- buttons. Personally I would like 30 mins for duration and 20% for percentage.

### Profile and target buttons on home screen

1. I think a bit more padding could be added as per my PDF. This will make them easier to press. 
2. I would really like the option to **reverse the press functionality** of the Profile button, so **short-press brings up the Profile Switch screen** and long-press brings up the profile info screen. I use the profile switch screen several times a day and having to long-press to access it is frustrating.

### Pop-up interface

There are UI issues with the pop-up interfaces for Profile switch and temporary target (and other pop-ups):

1. The "-/+" buttons need more padding and spacing. They're too difficult to press. Shown in the PDF.
2. Remove background transparency. You can't click on it to close the UI, so it is not needed. If we keep the transparency, we should be able to click the background area to close the popup. Shown in the PDF.
3. Interface slides around when keypad appears/disappears. Let's snap the whole UI to the top, or always show the keypad. Shown in PDF.