nightscout/AndroidAPS #354: NSClient 2.8.2 crashes during start

BernhardRo, 2021-02-20:
Hello,

I have installed NSClient 2.8.2, configured it to connect to my NS (which worked all). But a few seconds after the first values from NS come in, the App disappeared. New starts only showed the Android APS logo and the the app disappeared again. I had the notification of NSClient in the notification area, but it showed only something like "Getting values from Nightscout".
The data in Nightscout is my uploaded CGM data from XDrip (not from AAPS) as I do not have a pump, and I want to use NSClient to have the values on my smartwatch, which is much more convenient than with XDrip
I was able to debug NSClient and also run it in an virtual Android device with the same result (crashing). I traced the error in the app: the crash was caused by a Null pointer exception in 
src\main\java\info\nightscout\androidaps\interfaces\PumpDescription.java SetPumpDescription Line 108
the value of `pumpCapability` was null
I tried to find out where it should came from and found out that
src\main\java\info\nightscout\androidaps\plugins\pump\common\defs\PumpType.java
was always instantiated as 
`MDI("MDI", ManufacturerType.AndroidAPS, "MDI");` for which `pumpCapability` is null. (I guess instead of GenericAAPS as I do not have a pump at all)

I managed to replace the code for calling MDI with the code from GenericAAPS which works now for me, but I think this is only a quick and dirty hack which may work for me but is not a real solution.

Regards,
Bernhard

MilosKozak, 2021-02-20:
I don't undestand. NSClient is not using MDI but only VirtualPump and it's not possible to change it

BernhardRo, 2021-02-20:
Well I am not really in the details of the code and its dependencies. But for me the line
`src\main\java\info\nightscout\androidaps\plugins\pump\virtual\VirtualPumpPlugin.kt`
    ```
fun refreshConfiguration() {
        val pumptype = sp.getString(R.string.key_virtualpump_type, PumpType.GenericAAPS.description)
```
returned a pumptype of MDI type.

MilosKozak, 2021-02-20:
I'm unable replicate this issue. You can pm me your settings and master password privately

BernhardRo, 2021-02-22:
@MilosKozak: Were you able to reproduce it? I have send you my config per Email.

andyrozman, 2021-03-07:
MDI can never be selected, because its not a Pump. How were you able to select it?

BernhardRo, 2021-03-11:
Actually my problem was that nothing else was selected - I had to modify MDI to make it work. But I do not know why MDI was selected. Milos has my configuration. He should be able to reproduce it!

closed 2021-10-01 by MilosKozak