nightscout/AndroidAPS #358: Change AAPS 2.8.2 language will not chnage

robertrub, 2021-02-22:
I was testing for an English text in the French interface and fell on a bug. I changed the language from "System default" to German but everything stayed in French. 3 dot menu and quit AAPS. Restart AAPS and still, everything stayed in French.
[image]
[image]

Philoul, 2021-03-06:
Select System default in AAPS settings and change language in your android setting, then it works perfectly.
@MilosKozak I propose to remove this setting in AAPS (always set "system default") and show a reminder of the selected language in the android settings.

robertrub, 2021-03-06:
This was something that was working in 2.7 as, while replying to the questions for the objectives, I had changed the language in AAPS in order to see the questions in English for some of them.

Philoul, 2021-03-06:
It's a recurring bug that exists for long time (sometimes it works on some fields)
See 
- https://github.com/MilosKozak/AndroidAPS/issues/2759, 
- https://github.com/MilosKozak/AndroidAPS/issues/2746, 
- https://github.com/MilosKozak/AndroidAPS/issues/2152, 
- https://github.com/MilosKozak/AndroidAPS/issues/1060

robertrub, 2021-03-06:
Ok. As I said, not important. I wonder if anybody (except a very few) ever tried using it.

closed 2021-03-06 by robertrub

Philoul, 2021-03-06:
Mainly developers and translators I guess use it.
We all use now Android language switching with "Default System" in AAPS...