nightscout/AndroidAPS #370: Very high and increasing RAM usage the longer AAPS runs

RaduLupu1, 2021-02-24:
As the title says the RAM usage of the app starts with 200mb and starts increasing and increasing up to 1GB after the first day and continues until the app gets laggy and I have to restart it, all within 1 day.

I used an older version (2.6)

 for a long time, did an update and then the problem occured.

Built version: 2.8.2 fullrelease.
Smartphone: Redmi Note 7, MIUI 12.0.3

[AndroidAPS._2021-02-24_00-00-06_.0.zip](https://github.com/nightscout/AndroidAPS/files/6034173/AndroidAPS._2021-02-24_00-00-06_.0.zip)

Air69, 2021-02-24:
MIUI12 dexcom (g6, g5) cannot broadcast to AAPS, right?

guydavies, 2021-02-24:
That’s a function of the Dexcom patched app, not AAPS.