nightscout/AndroidAPS #380: AAPS SMB Mode: doesn't work with XDrip anymore - in latest Dev Versions

Alamo04, 2021-02-27:
From time to time I test the SMB Algo and the SMB options. Had never a problem to activate SMB with XDrip as BG source in the past. In the lasts dev version's XDrip, with Dexcom G6 Transmitter in Native Mode, was not longer accepted as BG Source for SMB

Are there some changes in the Dev versions, that XDrip will not longer work for SMB Mode ?

Did a lot tests to check whats wrong, but found nothing, so I think it's the Dev version that cause this problem.

**Test Versions:**
AAPS Master 2.8.2 - SMB works
AAPS DEV 2.8.2.x - SMB didn't work see Screenshot (it is in German, sorry)

**Also incl. the Log file.**

Normaly I use the Dark Mode Version from Rig22, but have tested it also with 3 different orig. Dev Versions, only the Master works.

[image]
[AndroidAPS_LOG_1614443052570.log.zip](https://github.com/nightscout/AndroidAPS/files/6054929/AndroidAPS_LOG_1614443052570.log.zip)

mick711, 2021-02-28:
Dith built e48200, 2.8.2.1dev b , omnipod with xdrip and g6 smb work for me

Alamo04, 2021-03-01:
@mick711
That is a thing that I can't understand, my version is also 2.8.2.1b, Medtronic with XDrip and Dexcom G6.

But if I want to use SMB, than AAPS reports (try to translate):
Safety: SMB will ever and after carbs deactivated,  because selected BG Source doesn't support trusted filtering of the values.

The Master Version will work fine with the same setting and SMB.

Also:
It looks like that there are many changes in xdrip plugin.

Tornado-Tim, 2021-03-01:
If you are using the G6 with Xdrip algorithim this is normal and has been for some time.

You need to run a native session to enable the full SMB stack.

Alamo04, 2021-03-01:
I never run the G6 without Native mode in XDrip, this wouldn't also not work in master.

swissalpine, 2021-03-02:
I see this safety comment too (xDrip with G6 native), but all SMB are delivered as expected ...

Alamo04, 2021-03-02:
@swissalpine 
SMB with Carbs and with TTs are working, if enabled.
SmB allways and SMB after Carbs not.

But this is a dev version, perhaps they are working on something.
Wanted only to be sure if that is a bug and let them know that. Or is it intend in this version, than everything was fine.

swissalpine, 2021-03-03:
Yes. I see it now. This seems to be buggy.

swissalpine, 2021-03-03:
Probably fixed with https://github.com/nightscout/AndroidAPS/pull/389

swissalpine, 2021-03-03:
At first moment I thought the fix won't help. But after the first new blood glucose value comes in, the security notice that SMB was disabled disappeared on SMB tab.

closed 2021-03-03 by MilosKozak

Alamo04, 2021-03-03:
@swissalpine 
confirmed, works now.  :)