nightscout/AndroidAPS #384: Pod fault 049 when AAPS tries set a zero temp basal and a time change event has just occured

fredrik-ljung, 2021-02-28:
AndroidAPS 2.8.2
Build version: c07465b
Date and time of the error: 28 February 2021 14:16:17

The phone used is a Samsung Galaxy S20 SM-G980F

Nothing done with the phone at the time. It just suddenly starts to scream and reports pod fault "A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected. Please deactivate your Pod and activate a new one".

When looking in the loggs it looks like this is the same error as I reported in https://github.com/AAPS-Omnipod/AndroidAPS/issues/40, which was corrected in the 2.8.0 release.

APPS sets a long zero temp basal, and during this period a time change event occures. When the zero temp basal expires AAPS tries to set a new temp basal, but this fails with a pod fault 49 error and the pod is lost. 

The 2.8.0 release worked fine, no lost pods, as long as the 'DST/Time zone detect' was enabled.

[AndroidAPS_LOG_1614518290876.log.zip](https://github.com/nightscout/AndroidAPS/files/6057811/AndroidAPS_LOG_1614518290876.log.zip)

A "Time, Date and/or TimeZone changed" event happens at 13:23:33 in the log.

[image]

fredrik-ljung, 2021-04-26:
I removed the code that prevents the time in the pod from beeing updated when a time change event is received and the time change is not a DST change or a time zone change. This means that the time on the pod is now allways updated when the time on the phone is updated.

[always_update_podtime.patch.zip](https://github.com/nightscout/AndroidAPS/files/6379204/always_update_podtime.patch.zip)

My girlfrend has now been using an AAPS 2.8.2 with this change for 6 weeks and has not lost a single pod during that period.

This change seams to solve the reported problem, however it is probably not a good solution for everyone since it generates a time zone change event in AAPS almost every time my girlfriend leavs the apartment.

samspycher, 2021-10-05:
@fredrik-ljung has this re-occured since your  patch?

fredrik-ljung, 2021-10-05:
The patch has worked fine. Only one screamer with pod fault 49 has occurred since the patch was applied 7 months ago.

samspycher, 2021-10-11:
@MilosKozak The removed code was implemented for the reasons mentioned in https://github.com/nightscout/AndroidAPS/pull/192

I have a WIP PR  https://github.com/nightscout/AndroidAPS/pull/726 which currently just removes the offending code as in the patch, but I am thinking about how to lower the frequency without preventing them altogether.

closed 2021-10-12 by MilosKozak