nightscout/AndroidAPS #391: SMB with high temp targets but working

mistermintsgh, 2021-03-03:
As the title says, I can't get SMBs to work with high temp targets.

They work fine for any TT of 5.5 or below, but despite the option for using SMB with a high TT being enabled, they just don't fire.

I have some automations running to catch highs and I don't necessarily want to set an overly aggressive TT. I have adjusted all of these automations to have a target of 5.5 or below, and manually set TT of above and below 5.5 to verify and can confirm on my setup at least, they definitely do not fire with a TT higher than 5.5

AAPS 2.8.2
Android 11
Samsung Galaxy S21 Ultra
Omnipod pump and RileyLink
[image]

 Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

Tornado-Tim, 2021-03-30:
Have a look at the OpenAPS SMB Tab, that will show you why SMB's are disabled despite setting the above settings.

mistermintsgh, 2021-03-30:
@Tornado-Tim Is there anything in particular I should be looking for on that tab? Any typical messages that should show?

I find it a bit odd that it will fire SMBs just fine when I set a target of, say, 5 mmol/L, but yet won't when aiming for the higher, and technically safer/not as near to hypo status 6 mmol/L

closed 2021-10-01 by MilosKozak