nightscout/AndroidAPS #392: Delta show and go back to 0

mick711, 2021-03-03:
I use dev 2.8.2.1.dev c built 41c4aa
I dont know if it's thé patch #389 but now Delta show correctly few second and go to 0 .

[image]
[image]

swissalpine, 2021-03-04:
This is usually the result of a wrong configuration: AAPS receives the same blood glucose value twice in quick succession, so the delta then inevitably becomes 0.
Check your settings, e.g. deactivate "Autobackfill BG".

Alamo04, 2021-03-04:
Confirmed.
Works until I update yesterday to the Dev, with your XDrip fixes, now Delta was also permanent 0.
Backfill missing values was enabled, disable them now and it works now - Delta is shown.

mick711, 2021-03-04:
Thanks swissalpine i disable autobackfile BG and work again .

closed 2021-03-04 by mick711