nightscout/AndroidAPS #394: Problem receiving warnings from Accu Chek Insight Pump

bertroode, 2021-03-04:
When my pomp issues an insulin low alert or a battery alert the smartphone beeps but no text message will be dislplayed.
 The alarm can only be silenced with the control panel on the pump.
After the alarm is received, communication with the pump remains on continuously. Also restarting AAPS will not disconnect the connection, only restarting the phone can solve the problem.
I am using a Samsung Galaxy S9+ with Android 10 and the Insight pump software V2.01.03.
This problem did not occur in previous versions of AAPS.

reijoahola-nightscout, 2021-03-04:
I have the same configuration, same pump version but Galaxy S9 / Android 10. I have set insulin level warning 20 U in the pump and the same level also in AAPS reservoir red status light. Typically I change the reservoir when reaching that 20 U limit. AAPS gives some notification/alarm in the phone but I have not noticed the problem you reported. I am not sure what alarm pump had time give in that situation but I expect that the notification that AAPS gives is based on the alarm from the pump.

bertroode, 2021-03-04:
In previous AAPS versions my phone showed a warning message and I was able to switch the alarm off with the phone.
Now the phone only beeps, but no message describing the kind of alarm and the communication connection with the pump is always on after the alarm has been triggered.

reijoahola-nightscout, 2021-03-04:
Do you have Android/AAPS Notifications on for "Insight Pump Alerts" ?

bertroode, 2021-03-04:
Yes, all warnings are on. The strange behaviour of the blue tooth connection worries me.

reijoahola-nightscout, 2021-03-05:
Today I tested again what happens if let Insight give reservoir level warning (W31). Now AAPS did not notice the warning and the pump gave an alarm about W31. I accepted the alarm from Insight pump and then it was OK.

bertroode, 2021-03-05:
The next time the insuline level triggers the alarm again I will save and analayze the log. The annoying consequence of this problem is that the pump's battery can quickly run out. That happened to me this week during the night. Because of my old age (76) I cannot hear the weak warning signal that sounds in such a situation.

reijoahola-nightscout, 2021-03-05:
I do not let the battery level go lower than 70%. I use in AAPS Status light (red) 70% and yellow is 95%. With that systen I have never reached the level where the pump gives battery alarm.

bertroode, 2021-03-06:
Well, quite a surprise what caused this problem. The documentation for the Insight pump states that only one communication channel should be set on the pump. However, the Insight pump has the option to set up 5 connections. If you remove the first row, the list won't scroll up in the display. So when three rows in the display are showing "--------" there still may be connections left  in  row 4 or 5. In my pump there was a connection for the hospita in row 5l. As a result, the alarms did not function properly and the connection with AAPS was also lost.

It may be usefull to add some extra information into the documentation of the Insight pump documentation  about stored connections in row 4 and/or 5.

closed 2021-03-06 by bertroode