nightscout/AndroidAPS #396: Feature request: Units left to inject while bolusing

robertrub, 2021-03-05:
When a bolus injection has started, the same info is given twice on the same window: the total of units to be injected.

It would be nice if the upper part gave the total of U **to be** injected and the lower part indicated the units **left** to inject.

[image]

Philoul, 2021-03-06:
Good Idea, I agree the proposal...

Philoul, 2021-03-08:
See screenshot of my dev phone (dev branch), it will be close to your proposal (Total in the upper part, not the unit left in the lower part but the injected Units)
[image]

I think you can close this issue.

robertrub, 2021-03-08:
Y§es, the translation is not good but we'll take care of that together ;)  x U ont étés injectées

closed 2021-03-08 by robertrub

Philoul, 2021-03-08:
I corrected translation