nightscout/AndroidAPS #406: Feature Request: Medtronic Bolus Progess

StephenBrown2, 2021-03-08:
As a user, I would like to see the estimated bolus progess properly fill the progress bar, blocking the UI while a bolus is in progress. This may help prevent errors due to attempting to perform extra pump-related actions such as setting a temp basal rate or adjusting the profile while the pump is busy.

I dug around in the code and found that the OmniPod driver is updating it, as well as the VirtualPump, and so I thought I might steal some code from either of them, however they are implemented very differently.
For my Medtronic 723 pump, I have found that the bolus delivery rate is the same as OmniPod, i.e. 0.025 units per second, so perhaps that code could be a good starting point.

I might be able to contribute the change myself, but I'd need to know if I'm headed in the right direction first.

MilosKozak, 2021-03-08:
MDT doesn't provide progress status like other pumps do

StephenBrown2, 2021-03-08:
No, and it appears OmniPod doesn't directly either, however it can be estimated based on the response received by the pump and a given bolus delivery rate.

StephenBrown2, 2021-03-08:
I'm thinking in particular of these lines in the OmniPod driver:

https://github.com/nightscout/AndroidAPS/blob/91e323b5f59634743513b2622139734651c78683/omnipod-eros/src/main/java/info/nightscout/androidaps/plugins/pump/omnipod/eros/manager/AapsOmnipodErosManager.java#L379-L389

https://github.com/nightscout/AndroidAPS/blob/91e323b5f59634743513b2622139734651c78683/omnipod-eros/src/main/java/info/nightscout/androidaps/plugins/pump/omnipod/eros/driver/manager/OmnipodManager.java#L358-L378

and, more simply, from the VirtualPumpPlugin: https://github.com/nightscout/AndroidAPS/blob/91e323b5f59634743513b2622139734651c78683/app/src/main/java/info/nightscout/androidaps/plugins/pump/virtual/VirtualPumpPlugin.kt#L189-L197

with, perhaps, pumpDescription.bolusStep set to 0.05 and pumpDescription.bolusStepDuration set to 2000 milliseconds, adding this loop:
```kotlin
        var delivering = 0.0
        while (delivering < detailedBolusInfo.insulin) {
            SystemClock.sleep(pumpDescription.bolusStepDuration)
            val bolusingEvent = EventOverviewBolusProgress
            bolusingEvent.status = resourceHelper.gs(R.string.bolusdelivering, delivering)
            bolusingEvent.percent = min((delivering / detailedBolusInfo.insulin * 100).toInt(), 100)
            rxBus.send(bolusingEvent)
            delivering += pumpDescription.bolusStep
        }
```
to right around here: https://github.com/nightscout/AndroidAPS/blob/91e323b5f59634743513b2622139734651c78683/medtronic/src/main/java/info/nightscout/androidaps/plugins/pump/medtronic/MedtronicPumpPlugin.java#L855-L858
might do the trick?

andyrozman, 2021-03-08:
The main problem is that there is no easy way to cancel bolus (only way to cancel bolus on MDT is to put pump in suspend), which is why this was never implemented with dialog. And with difference to other pumps, bolus takes a long time on MDT, so displaying dialog for 15 minutes might be a problem (it takes 42s for 1U to be delivered).

Every other application does it the same way. 

We do block pump driver in the meantime, so that no action can be done during this time.

StephenBrown2, 2021-03-08:
> there is no easy way to cancel bolus (only way to cancel bolus on MDT is to put pump in suspend)

Agreed. The Cancel button could be disabled, but I would think that the current alarm message in `strings.xml` would still suffice.
```
85:    <string name="medtronic_cmd_cancel_bolus_not_supported">You cancelled Bolus, after it was already set on Pump. Since Medtronic Pump doesn\'t support cancel, you will need to manually cancel it. Put the Pump into Suspend mode and then do Resume (if you still want to cancel). Application will pick up changes, on next update (in less than 5 minutes).</string>
```

> And with difference to other pumps, bolus takes a long time on MDT, so displaying dialog for 15 minutes might be a problem (it takes 42s for 1U to be delivered).

I am not familiar with other pumps (Aside from the Medtronic 670G and Animas pumps, which did have a "Fast Bolus" option). How long does it take an OmniPod to deliver 1U?

> Every other application does it the same way.

I'm also unfamiliar with any other application providing remote bolus for Medtronic pumps. Can you provide some examples?

> We do block pump driver in the meantime, so that no action can be done during this time.

This is what would cause me grief, thinking I could bolus and then immediately set a temp target or adjust my profile to handle extended carbs or anything else that might need pump communication, when it would eventually time out due to the pump being busy, so I would need to pull out my pump and watch it for when it would actually finish.
Blocking the UI would prevent that from happening, and those that don't need it wouldn't be looking at the screen anyways.

Webifi, 2021-03-10:
> Blocking the UI would prevent that from happening, and those that don't need it wouldn't be looking at the screen anyways.

I'd rather not have the UI blocked -- even the blocking hat happens while waiting for the pump is annoying.   A queue for pump commands would be nice in this case -- with warning before items are placed in the queue, and the ability to delete items from the queue before they get sent to the pump.

Techdim, 2025-04-23:
Will this requirement continue to be developed later? I just switched from the Loop, and for Medtronic 722 , Loop can display progress and can cancel the bolus 
AAPS 3.3.2.0 
Android 14 
Xiaomi HyperOS 1.0.12

andyrozman, 2025-04-23:
No. It was never planned to be done. 

Like I said before we have two problems:
- very long time needed to bolus (42s / 1U)
- no direct way to cancel bolus (unless you want to suspend pump, which would also break all currently running TBRs)

If someone decides to implement this (which I wouldn't recommend), it needs to be implemented in such way that functionality can be turned off by user (in settings). Not everybody will welcome his AAPS having blocked for x minutes.

closed 2026-10-15 by MilosKozak