nightscout/AndroidAPS #411: Minimal request change active during closed loop

larsqvd, 2021-03-10:
Tiny change, but minimal change request setting is active during closed loop operations even though it has no effect and should not be used. 

[image]

andyrozman, 2021-03-11:
This setting is always used... And its logical that its here... It tells how much the value must change, to be sent to pump. So for example, If your TBR is 120% and on next calculation requested TBR would be 130%, this change won't be sent to pump, until requested change is over 15%... So when AAPS determines that TBR needs to be 140%, then it will be sent to pump. Default value is 20% though... If your pump supports increment by 10, having value at 15 is useless...

larsqvd, 2021-03-11:
Ah thanks for explaining Andy, I will check for adapting the guide then as it is described as open loop only feature and because of my 15% it was not functional in my closed loop ;-)

closed 2021-03-11 by larsqvd