nightscout/AndroidAPS #418: AndroidAPS restarts on bad blob of historical data from Medtronic pump

Amunhateb, 2021-03-15:
I am having an issue with AAPS restarting every 7-9 minutes. Judging by logs it is related to historical data incorrect date value. I've attached the log file. Issue could be found by looking up Exception.

[AndroidAPS._2021-03-04_15-24-25.log.zip](https://github.com/nightscout/AndroidAPS/files/6140163/AndroidAPS._2021-03-04_15-24-25.log.zip)

Amunhateb, 2021-03-15:
@andyrozman here you go, thx for finding out the cause

MilosKozak, 2021-03-17:
https://console.firebase.google.com/project/androidaps-c34f8/crashlytics/app/android:info.nightscout.androidaps/issues/297f1f2deabd18e680a04b3f5cf33672?versions=2.8.2%20(1500)&time=last-twenty-four-hours&sessionEventKey=60517F110369000151DE4ED6E5E43AEB_1518914798545604254

andyrozman, 2021-03-17:
@Amunhateb Fix is now done in this branch, please install and test if this fixes everything... I have run it through my history decoded and all entries seem to be correct now. Please write here so that I can then do PR so that it can get released in next version. 
https://github.com/andyrozman/AndroidAPS/tree/bug418_historyFix_BolusWizardRecordProblem

Amunhateb, 2021-03-17:
@andyrozman I've tested it. Everything is good, thx.

duzy78, 2021-03-18:
I am experiencing similar problem but it looks that different field is problematic (dayOfMonth)

[AndroidAPS_2.8.2.1-dev-build.log](https://github.com/nightscout/AndroidAPS/files/6163704/AndroidAPS_2.8.2.1-dev-build.log)

Should I log in as a separate problem or should it be fixed along with this one?
The log is from the branch with potential fix for problems with parsing historical data.
If you need more data please let me know.

andyrozman, 2021-03-18:
@duzy78 Create separate ticket please... While the end result seems to be the same (that date can't be parsed), this doesn't mean that there is a same base problem...

andyrozman, 2021-03-20:
PR was opened: https://github.com/nightscout/AndroidAPS/pull/430 and merged into dev. Closing this issue.

closed 2021-03-20 by andyrozman

Amunhateb, 2021-03-21:
> PR was opened: #430 and merged into dev. Closing this issue.

https://github.com/nightscout/AndroidAPS/pull/422 is the correct PR. Thanks for fixing it!

andyrozman, 2021-04-14:
@Amunhateb I have created special hotix branch that is based on master and contains all 3 medtronic fixes: https://github.com/andyrozman/AndroidAPS/tree/hotfix_mdt_2_8_2