nightscout/AndroidAPS #420: [Enhancement] Can you please bring in a battery status for OrangeLink into the Medtronic driver ?

Alamo04, 2021-03-16:
Would be nice to see the status,  without delete OrangeLink in AAPS and scan it with nRF Connect.

andyrozman, 2021-03-17:
@Alamo04 At the moment if you enable battery display in settings, you can see battery level in "RileyLink Stats" dialog.

I will add display of battery level on main MDT tab too, with this ticket.

Alamo04, 2021-03-17:
@andyrozman 
I have no RileyLink Stats Dialog page, only a RileyLink Configuration page and there I can only connect/disconnect a RileyLink. I think the page you are talking about is in the Omnipod driver integrated ?

But I'm happy, that you can bring it to the MDT tab and for your realy impressiv fast response. 🙂

StephenBrown2, 2021-03-17:
It's in the MDT tab, on the bottom right, circled here.

Alamo04, 2021-03-18:
Oh god, never noticed this button. 🥴
Thanks @StephenBrown2

StephenBrown2, 2021-03-18:
No prob, you're welcome! 👍🏻

bchris21, 2021-03-19:
For me for a strange reason RL battery 🔋status on RL Stats is always 100% even after 2 weeks of continuous use. I no longer check it. Is there any way I can help to check the issue? I am using Energizer Lithium for my Orangelink.

Alamo04, 2021-03-19:
@bchris21
Thats what Lithium batteries does. They lasts very long and hold their power over a long time.
But if they begin to drop their power, then it comes really fast to an end.

I read, that some have alkaline batteries in use and are still at 50% after 5 weeks.

So don't worry,  be happy.  😉

Alamo04, 2021-03-19:
@bchris21
For info:
Someone wrote, 5 minutes before, that his Lithium batteries lasts now 53 days and that they today dropped to 90%.

andyrozman, 2021-03-20:
Opened PR for this: https://github.com/nightscout/AndroidAPS/pull/430

closed 2021-03-20 by MilosKozak