nightscout/AndroidAPS #423: AndroidAPS is stopped when processing Medtronic pump history

duzy78, 2021-03-18:
It has happened on AndroidAPS 2.8.2. It is 100% reproducible since it started, so AAPS is being restarted every time pump sends history. 
Phone: Samsung Galaxy S7 (Android 8.0)
Pump: Medtronic 715 (firmware 2.3A)
RileyLink

After changing battery in insulin pump, application started to stop during retrieving historical data from insulin pump.
From log it looks that parsing of history is causing the problems:
org.joda.time.IllegalFieldValueException: Value 0 for dayOfMonth must be in the range [1,31]

[AndroidAPS_2.8.2_history_parsing_problem.log.zip](https://github.com/nightscout/AndroidAPS/files/6163923/AndroidAPS_2.8.2_history_parsing_problem.log.zip)

andyrozman, 2021-03-31:
PR was merged, so this is fixed... Problem was BGReceived512 entry... It should work correctly now...

andyrozman, 2021-03-31:
https://github.com/nightscout/AndroidAPS/pull/448

andyrozman, 2021-04-14:
Since this was merged into dev we can close it...

closed 2021-04-14 by andyrozman

andyrozman, 2021-04-14:
@duzy78 I have created special hotix branch that is based on master and contains all 3 medtronic fixes: https://github.com/andyrozman/AndroidAPS/tree/hotfix_mdt_2_8_2