nightscout/AndroidAPS #435: AAPS stops running loop silently (no errors)

twain47, 2021-03-21:
Occasionally AAPS stops running loop. Autosens also stops updating. AAPS log shows:

```21:40:59.051 [Thread-659] D/AUTOSENS: [IobCobCalculatorPlugin.createBucketedData5min():456]: Adjusting bucketed data time. Current: 21/03/21 20:15:10 to: 21/03/21 20:13:54 by 76 sec
21:40:59.051 [Thread-659] D/AUTOSENS: [IobCobCalculatorPlugin.createBucketedData5min():456]: Adjusting bucketed data time. Current: 21/03/21 20:20:38 to: 21/03/21 20:18:54 by 104 sec
21:40:59.051 [Thread-659] D/AUTOSENS: [IobCobCalculatorPlugin.createBucketedData5min():459]: Fallback to non 5 min data
21:40:59.052 [Thread-659] D/CORE: [IobCobCalculatorPlugin.createBucketedDataRecalculated():365]: Adjusted time 21/03/21 21:41:18
21:40:59.052 [Thread-659] D/AUTOSENS: [IobCobOref1Thread.run():126]: Aborting calculation thread (No bucketed data available): onEventNewBG
21:40:59.052 [Thread-659] D/AUTOSENS: [IobCobOref1Thread.run():396]: AUTOSENSDATA thread ended: onEventNewBG
```

In this case it seems to be something to do with how eversense logs BG data but it may impact other sensors.

Increasing the fallback threshold slightly allows loop to continue working.

MilosKozak, 2021-03-24:
what version?

overfrenk, 2021-03-24:
2.7.x to 2.8.x all version

overfrenk, 2021-05-15:
Will there be something on the new versions that fixes this problem?

MilosKozak, 2021-06-23:
this is completely reworked in dev

closed 2021-06-23 by MilosKozak