nightscout/AndroidAPS #436: [2.8.2.1-dev-e] crashes after start

swissalpine, 2021-03-22:
Logcat:
```
2021-03-22 19:58:38.279 27533-27582/? E/AndroidRuntime: FATAL EXCEPTION: RxCachedThreadScheduler-1
    Process: info.nightscout.androidaps, PID: 27533
    io.reactivex.exceptions.OnErrorNotImplementedException: The exception was not handled due to missing onError handler in the subscribe() method call. Further reading: https://github.com/ReactiveX/RxJava/wiki/Error-Handling | java.lang.IllegalStateException: Migration didn't properly handle: foods(info.nightscout.androidaps.database.entities.Food).
     Expected:
    TableInfo{name='foods', columns={pumpId=Column{name='pumpId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, subCategory=Column{name='subCategory', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, gi=Column{name='gi', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, carbs=Column{name='carbs', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, isValid=Column{name='isValid', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, portion=Column{name='portion', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, nightscoutSystemId=Column{name='nightscoutSystemId', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, version=Column{name='version', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, pumpType=Column{name='pumpType', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, referenceId=Column{name='referenceId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, unit=Column{name='unit', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=0, defaultValue='null'}, nightscoutId=Column{name='nightscoutId', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, dateCreated=Column{name='dateCreated', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, endId=Column{name='endId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, pumpSerial=Column{name='pumpSerial', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, startId=Column{name='startId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, protein=Column{name='protein', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, name=Column{name='name', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=0, defaultValue='null'}, fat=Column{name='fat', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, id=Column{name='id', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=1, defaultValue='null'}, category=Column{name='category', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, energy=Column{name='energy', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}}, foreignKeys=[ForeignKey{referenceTable='foods', onDelete='NO ACTION', onUpdate='NO ACTION', columnNames=[referenceId], referenceColumnNames=[id]}], indices=[Index{name='index_foods_referenceId', unique=false, columns=[referenceId]}]}
     Found:
2021-03-22 19:58:38.279 27533-27582/? E/AndroidRuntime: TableInfo{name='foods', columns={pumpId=Column{name='pumpId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, subCategory=Column{name='subCategory', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, gi=Column{name='gi', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, carbs=Column{name='carbs', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, isValid=Column{name='isValid', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, portion=Column{name='portion', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, nightscoutSystemId=Column{name='nightscoutSystemId', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, version=Column{name='version', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, pumpType=Column{name='pumpType', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, referenceId=Column{name='referenceId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, unit=Column{name='unit', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=0, defaultValue='null'}, nightscoutId=Column{name='nightscoutId', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, dateCreated=Column{name='dateCreated', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, endId=Column{name='endId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, pumpSerial=Column{name='pumpSerial', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, startId=Column{name='startId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, protein=Column{name='protein', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, name=Column{name='name', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=0, defaultValue='null'}, fat=Column{name='fat', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, id=Column{name='id', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=1, defaultValue='null'}, category=Column{name='category', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, energy=Column{name='energy', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}}, foreignKeys=[ForeignKey{referenceTable='foods', onDelete='NO ACTION', onUpdate='NO ACTION', columnNames=[referenceId], referenceColumnNames=[id]}], indices=[]}
        at io.reactivex.internal.observers.EmptyCompletableObserver.onError(EmptyCompletableObserver.java:50)
        at io.reactivex.internal.operators.completable.CompletablePeek$CompletableObserverImplementation.onError(CompletablePeek.java:95)
        at io.reactivex.internal.operators.completable.CompletableSubscribeOn$SubscribeOnObserver.onError(CompletableSubscribeOn.java:74)
        at io.reactivex.internal.operators.completable.CompletableFromCallable.subscribeActual(CompletableFromCallable.java:40)
        at io.reactivex.Completable.subscribe(Completable.java:2309)
        at io.reactivex.internal.operators.completable.CompletableSubscribeOn$SubscribeOnObserver.run(CompletableSubscribeOn.java:64)
        at io.reactivex.Scheduler$DisposeTask.run(Scheduler.java:608)
        at io.reactivex.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:66)
        at io.reactivex.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:57)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
```

Philoul, 2021-03-22:
fixed in latest dev (https://github.com/nightscout/AndroidAPS/commit/e762efeaa6a80b3b70930d6c12906162f860ed17)
Can you update your dev branch and try again ?

swissalpine, 2021-03-23:
Yes. That fixes the crashes. Thanks for your fix!

closed 2021-03-23 by swissalpine