nightscout/AndroidAPS #449: DST: Pump connection lost

nshtg, 2021-03-29:
Version 2.8.2

This year during the time change AAPS issued an alarm exactly at 03:00 due to a possible loss of connection to the pump. The limit entered in the options was 30min. I suspect that here simply the time since the last connection was checked and after the time change of course suddenly a whole hour was in between.

Milahan, 2024-03-31:
Exactly the same issue here tonight (v3.2.0.3, with Dana I). I hope this can be fixed, as suggested by nshtg.

MilosKozak, 2026-10-15:
Fixed in ea6898ea1c (dev).

The cause was not the time since the last connection. That is counted in UTC milliseconds and does not jump when the clock changes.

When the clock goes forward, the local hour from 02:00 to 03:00 does not exist. Dana RS and Dana-i read the pump clock as a local time, and the time library throws an error for a time in that hour. The reply then counted as not received, AAPS reset the pump, and its last connection time became 0. So "pump unreachable" came at once, whatever limit was set.

- Dana RS (the pump keeps local time): right after the change the pump clock still shows the missing hour, so the reads failed for about one hour from 03:00.
- Dana-i (the pump keeps UTC): the UTC clock was also read as a local time, so the reads failed for one hour from 04:00 (in central Europe).

Now the Dana-i clock is read as UTC, and a local time in the missing hour is read as the next hour, which is the real time while the pump clock is behind. The same for the pump history and dates. The old DanaR driver already handled this.

closed 2026-10-15 by MilosKozak