nightscout/AndroidAPS #450: combination of (AND + OR) is not possible in automation

TranceCake, 2021-03-30:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem:
  I am trying to set this event in automation: IF ((bg >= 7 AND delta >= 0.3) OR bg >= 10)

[image]

[image]

[image]

[image]

So only the outer AND/OR choice seems to work and it sets all operators and not just the one on the outer clause

- Build Version: 
  AAPS 2.8.2
  14559a01f-2021.03.25-12:57

- Logs:
  Will add shortly when I have access to a computer

robertrub, 2021-03-31:
I confirm. I couldn't create a mix of *and* + *or* conditions.

TranceCake, 2021-05-15:
it was brought to my attention that you can achieve this if you first select all condition and then fill in the values. This is not intuitive, it took me 3 tries to understand it. This could maybe use a UX pass, to which I'm happy to contribute!

closed 2021-05-15 by TranceCake