nightscout/AndroidAPS #456: [All lasts Dev's] AAPS entrys are deleted  in NightScout next morning

Alamo04, 2021-04-02:
My system:
AAPS latest Dev, Pixel 5 Phone, Medtronic 754, OrangeLink.

Time of issue: Since a month, or longer, with different Dev versions.

Issue:
Canula-, Reservoir-, Battery- and Sensorchange, all of them are entered in AAPS from hand, or over pump history.
AAPS load this changes up to NightScout and everything is fine.
Next morning, all this changes are disappeared in NightScout and the old values are shown. This doesn't happen in AAPS.

I have to enter all this changes from hand into NightScout, or have to do it with AAPS again, with using date and time of the changes from the day before, then they lasts in Nightscout.

Can't attache log file, because my tablet and Smartphone doesn't download zip files.
So I mailed one to Andy Rozman. Perhaps I can attach it later with my pc.

PS: 
I wonder that it seems, that I'm alone with this issue, because there is no open one.
My Database at Atlas is pretty empty, so this is not the case.

swissalpine, 2021-04-03:
No, I can confirm this issue.

andyrozman, 2021-04-04:
A lot of data is not stored into Nightscout... While you can go into stats and see pump history, not everything in pump history is stored in Nightscout.... Most of items are only in App history (not even in app database) and are reloaded when you start AAPS next. What we store is: Boluses, TBRs, Cannula change and Insulin.Reservoir change. Everything else is ignored. On every start we read history for 36 hours (so we usually read 1-3 history pages), and that history is cleaned every hours, so you can see only 36 hours of history... That is when you go into RL Stats window and use the last tab...

Alamo04, 2021-04-04:
@andyrozman 
Thats ok, but Canula- and Reservoir change are deleted in Nightscout after a few hours, also batterie.- and sensor change.
In the past, AAPS stored this information in Nightscout and everything was fine. 

Since a month, or longer this entry's are deleted over night.

I change my Sensor-, Canula- and Reservoir normally at evening, Batterie every 14+ days over the day after she was died. 
All of this changes are shown correct in Nightscout, over the hole day and are deleted next day.

And that has nothing to do with history read, this happens also if I use the action Tab for batterie- and Sensor change.
So it has nothing to do with history readings, or pump driver.

That's a core AAPS issue.

andyrozman, 2021-04-04:
OK. Since first pump mentioned was Medtronic, I though this was MDT problem, but yes you are right, most of this event are not triggered from pump... So it can't be my fault... 
Guys are still changing database AND nightscout client, so that might have to do something with that... Dev shouldn't be used for looping, until its stable and released.

Alamo04, 2021-04-04:
I know that Dev versions are not stable and can have multiple issues.

I have 4 Medtronic pumps and 3 old and one new phones. 2 NightScout sites.
I use one phone for looping and one for testing 24/7.

The testing system is only for finding issues in dev versions and report them, but perhaps you are right and it is not helpful to do so.

I will stop testing dev versions and will do that only for Release Candidates and Master in the future.

swissalpine, 2021-04-04:
Here you can clearly see: In the past, data was saved, now it is not.
By the way: If nobody tests dev, not all bugs will be found. ;-)
[image]
[image]

swissalpine, 2021-04-12:
Canula age and Insulin age seems to work now ... - without any changes.

[image]

Alamo04, 2021-04-12:
@swissalpine 
I enter all this data now in AAPS first and then also in NightScout, to be sure that they don't be deleted.
Will test it again with next Dev, but I'm still not sure if that is a bug. Perhaps it is intend ?

Don't know what to do, should I close this issue, because it is a dev version. Let Milos take this choise.

@MilosKozak
feel free to close this issue, if it is not helpful, or the work in progress is the reason.

MilosKozak, 2021-06-23:
hopefully fixed by https://github.com/nightscout/AndroidAPS/commit/a70f24a68aca5d98300c6e622e7732a3b833e426

closed 2021-06-23 by MilosKozak